dnp3/opendnp3 #121: Segfault when calling MasterAdaptor.Disable() or MasterAdaptor.Shutdown()

damonsensus, 2015-02-04:
I started following the references to determine what the issue is exactly.  Not being familiar with the code, it might take some time, so I am posting this in the meantime.

I built from 2.0.x following the most recent release (2.0.1.1 based on the CLRInterface AssemblyInfo).

{System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
   at Automatak.DNP3.Adapter.MasterAdapter.Disable()
   at Sensus.DA.ScadaMaster.Dnp3.Outstation.Disconnect() in c:\develop\Sensus.DA.ScadaMaster\Sensus.DA.ScadaMaster\Dnp3\Outstation.cs:line 247
   at Sensus.DA.ScadaMaster.Dnp3.ChannelManager.InitChannel(ChannelCfg cfg) in 
... (elided as not relevant) ...
}

jadamcrain, 2015-02-04:
Ok. Can you forward this to the list? We'll discuss there.

jadamcrain, 2015-04-04:
Closing. Down the road we might might these calls idempotent.

closed 2015-04-04 by jadamcrain