dnp3/opendnp3 #122: Crash on trying to add event to event buffer full of other event types

atmurray, 2015-02-18:
In EventBuffer.h
During a call to EventBuffer::UpdateAny(const Event<T>& evt), when the outstation events is full of events of another type, the call to:
RemoveOldestEventOfType(T::EventTypeEnum)
fails and returns false. 

As a result, the subsequent call to:
events.Add(SOERecord(evt.value, evt.index, evt.clazz, evt.variation))
returns a NullPtr and so:
events.Add(SOERecord(evt.value, evt.index, evt.clazz, evt.variation))->value.Reset();
results in a crash.

Suggested fix:
if (!RemoveOldestEventOfType(T::EventTypeEnum)) return;

neilstephens, 2015-02-18:
The max size of the linked list that holds the events is initialised to the sum of the number of events before each type should overflow, so if you've configured the number of events for each type to something non-zero, the bug must be somewhere else. That suggested fix would still be good to implement for the sake of catching the zero length event buffer case, and probably log a warning.

If a zero length event buffer isn't the culprit, then something isn't overflowing properly, or maybe a problem with the linked list implementation?

jadamcrain, 2015-02-18:
Sounds like this would be easy to write a unit test for to isolate and fix.

Always easier to test the component directly than through a full integration. I can have a shot at it when I wake up a bit ;).

neilstephens, 2015-02-20:
Hi Adam,

Taking a step back... This issue arose within our data concentrator (which
we're working on open sourcing ), during a stress test, that could have
been causing concurrent access to a single outstation.

We had made an assumption that the transaction interface was thread safe,
but were we wrong? Are we meant to serially access the outstation stack?

We haven't seen any issues up until now, but this is the first time we've
pushed the limits of async io.

Cheers,
Neil.

jadamcrain, 2015-02-20:
Hi Neil,

IDatabase is absolutely intended to be thread-safe. The implementation uses pretty much the only explicit lock in the entire stack:

https://github.com/automatak/dnp3/blob/2.0.x/cpp/libs/opendnp3/src/opendnp3/outstation/Database.cpp#L47

I'm p rettyconfident it's just an issue that can be replicated with a single-threaded unit test.

-Adam

jadamcrain, 2015-02-20:
Looking at the code, something has definitely gone wrong with either 1) the list or 2) the state tracking for how many events there are of every type.

Can you guys get me detailed information on what the list contains vs what the "total counts" says when this condition occurs?

I would modify:

auto pNode = events.Add(SOERecord(evt.value, evt.index, evt.clazz, evt.variation));
if(pNode)
{
  totalCounts.Increment(evt.clazz, T::EventTypeEnum);
}
else
{
 // this shouldn't be happening. print diagnostics and exit on the list and the total tracking.
}

neilstephens, 2015-02-20:
Sorry Adam, we sent you on a bit of a wild goose chase.

Alan changed some code (yes, I'm dobbing you in Alan) to access the
outstation database directly, instead of using the transaction interface,
at the same time as moving to the new stack.

So it was our code causing the issue. Sorry again for letting you chase a
non-existent bug. Thanks for being so responsive. I hope you get this
message before doing any more work on this.

Cheers, Neil.

jadamcrain, 2015-02-20:
Hahaha. Alan now owes me a beer (Or a data concentrator). =)

I really didn't spend more than 15 minutes. I'm just happy there isn't a glaring issue.

Perhaps there's something we can learn from this to improve the API.

neilstephens, 2015-02-20:
A bit more detail of what went wrong:

A changed was made to use Database::Modify instead of using a
opendnp3::Transaction which calls Database::Start and Database::End
automatically. So Start and End weren't being called around the database
access.

Cheers, Neil.

jadamcrain, 2015-02-20:
OK. We should have a transaction flag and assertions that it's active. That would make things fail hard and fast as opposed to subtle/racy failures.

jadamcrain, 2015-02-20:
Yet another thing we could do:

Make Update/Modify/etc private. Transaction can be a friend class with its own mirror of the API. That would definitely fix this issue.

neilstephens, 2015-02-21:
yeah, that would be awesome if it was all wrapped up in the Transaction
class. I don't see a down side if all the methods are mirrored with the
additional automatic Start and End.

Cheers,
Neil.

neilstephens, 2015-02-21:
Although, if you're only looking to do small updates in release candidate
state, the simple addition of the assert you mention would do the job for
now.

Cheers,
Neil.

atmurray, 2015-02-21:
Unfortunately I don't think this was the only issue as I'm still getting the same crash when stress testing our datacon. Unless I'm not understanding what needs to be protected, after looking at all the calls on eventBuffer I suspect it is unprotected from concurrency in the following three locations:
```
OutstationContext::SetOffline()
OutstationUnsolicitedStateConfirmWait::OnConfirm(OutstationContext* pContext, const APDUHeader& header)
OutstationUnsolicitedStateConfirmWait::OnConfirmTimeout(OutstationContext* pContext)
```
I've added Transactions to these three function calls in my copy of the stack to see if this fixes things up and will let you know if it doesn't. I'm not exactly sure how to write unit tests to test for this sort of concurrency issues.

Thoughts?

jadamcrain, 2015-02-21:
I have an idea of how to fix this and get rid of the explicit locks completely. Let me know how the experiment goes, but I'm going to prototype a fix that makes the locks not needed anymore.

jadamcrain, 2015-02-21:
Ok guys.

Check out my prototype in the "measfix" branch. The main libs and demos are compiling so you should be able to build your linux binaries. If this works I'll update the tests and .NET:

https://github.com/automatak/dnp3/tree/measfix

I decided that trying to get all the locks correct all the time is a fools errand and even if we do get it right it will just cause regressions later.  Everything else in opendnp3 uses the execution context (IExecutor) to send messages to the correct underlying strand in a thread-safe way. Why shouldn't measurement updates from any thread also do this?

So now, instead of locking shared datastructures when we want to update things, we build a changeset that is dynamically allocated and applied automatically on the correct strand. The magic happens here:

https://github.com/automatak/dnp3/blob/measfix/cpp/libs/asiodnp3/src/asiodnp3/MeasUpdate.cpp#L46

The external API for using this thing is almost identical to before, and is much harder to use incorrectly:

https://github.com/automatak/dnp3/blob/measfix/cpp/examples/outstation/DemoMain.cpp#L112

The methods that MeasUpdate uses on IOutstation are private and MeasUpdate is a friend class.

Please try this branch out with your stress tests and let me know how it goes.

atmurray, 2015-02-21:
Stress test underway, so far so good (it usually crashes reasonably quickly).
I like the new asiodnp3::MeasUpdate and making hidden things outside the stack that shouldn't be touched.
Completely agree that strands seem to be the way to achieve thread-safety, makes things thread safe automatically which as you say avoids the risk of regressed bugs.

Will let you know how we go with further testing!
If this is all good, would you be leaning towards an RC-2 release of 2.0.1?

jadamcrain, 2015-02-21:
Yeah, we'll definitely issue RC2 pretty shortly after you confirm there are no more issues.

neilstephens, 2015-02-22:
I really like the new solution of exposing the strand internally for the
update interface.

I've done a linux build, and kicked off another test:

One event stream, asynchronously multiplexed 4000 ways, then demultiplexed
back to a single slave. It's humming along with no crashes so far.

I'll check back in a few hours.

Cheers,
Neil.

neilstephens, 2015-02-22:
2GB worth of events funneled into a single outstation in bursts of 4000 concurrent events every 100ms - passed no problems.

jadamcrain, 2015-02-22:
Ok, sounds like we're confident that multi-threaded access to event buffer was the root cause. I will close the ticket out once this gets merged into 2.0.1-RC2

jadamcrain, 2015-04-04:
Assuming we squashed this issue so closing,

closed 2015-04-04 by jadamcrain