dnp3/opendnp3 #129: build fail on ubuntu 14.04.2  on branch 2.0

tweakmy, 2015-07-19:
Any idea?
````
libtool: link: g++  -fPIC -DPIC -shared -nostdlib /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/4.8/crtbeginS.o  cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-AnalogCommandEvent.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-AnalogOutput.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-APDUBuilders.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-APDUHandlerBase.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-APDUHeader.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-APDUHeaderParser.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-APDULogging.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-APDUParser.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-APDURequest.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-APDUResponse.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-APDUWrapper.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-AppControlField.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-BinaryCommandEvent.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-BitReader.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-ClassField.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-ControlRelayOutputBlock.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-EventTriggers.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-FunctionHelpers.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-GroupVariationRecord.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-IINField.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-MeasurementTypes.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-ObjectWriter.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-OctetData.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-TimeAndInterval.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-ChannelState.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-CommandStatus.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-ControlCode.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-DoubleBit.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-FunctionCode.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-GroupVariation.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-IntervalUnits.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-LinkFunction.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-PointClass.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-QualifierCode.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-TaskCompletion.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-TimeSyncMode.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-CRC.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-IOpenDelayStrategy.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-LinkFrame.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-LinkHeader.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-LinkLayer.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-LinkLayerParser.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-PriLinkLayerStates.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-SecLinkLayerStates.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-ShiftableBuffer.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-AssignClassTask.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-ClearRestartTask.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-WriteTask.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-StartupIntegrityPoll.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-CommandTask.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-IMasterState.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-Master.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-MeasurementHandler.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-CommandResponse.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-EnableUnsolicitedTask.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-CommandMarshaller.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-MasterTasks.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-EventScanTask.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-IMasterTask.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-MasterParams.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-MasterScheduler.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-PollTaskBase.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-AssignClassTask.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-SerialTimeSyncTask.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-MasterContext.o cpp/libs/opendnp3/src/opendnp3/master/.libs/libopendnp3_la-ITaskLock.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-CRC.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-LinkHeader.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-SecLinkLayerStates.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-LinkLayer.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-ShiftableBuffer.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-IOpenDelayStrategy.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-LinkFrame.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-LinkLayerParser.o cpp/libs/opendnp3/src/opendnp3/link/.libs/libopendnp3_la-PriLinkLayerStates.o cpp/libs/opendnp3/src/opendnp3/transport/.libs/libopendnp3_la-TransportLayer.o cpp/libs/opendnp3/src/opendnp3/transport/.libs/libopendnp3_la-TransportRx.o cpp/libs/opendnp3/src/opendnp3/transport/.libs/libopendnp3_la-TransportTx.o cpp/libs/opendnp3/src/opendnp3/transport/.libs/libopendnp3_la-TransportStack.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-ChannelState.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-QualifierCode.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-DoubleBit.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-TaskCompletion.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-TimeSyncMode.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-CommandStatus.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-PointClass.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-ControlCode.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-FunctionCode.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-GroupVariation.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-LinkFunction.o cpp/libs/opendnp3/src/opendnp3/gen/.libs/libopendnp3_la-IntervalUnits.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group21.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group41.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group51.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group12.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group3.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group50.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group30.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group32.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group42.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group1.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group2.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group40.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group20.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group13.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group22.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group23.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group10.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group43.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group11.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group52.o cpp/libs/opendnp3/src/opendnp3/objects/.libs/libopendnp3_la-Group4.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-APDUHeader.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-OctetData.o cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-BitReade [... cut ...]
cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-AnalogOutput.o:/dnp3/cpp/libs/opendnp3/src/opendnp3/app/AnalogOutput.h:45: first defined here
cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-AnalogOutput.o: In function `opendnp3::AnalogOutputInt32::operator==(opendnp3::AnalogOutputInt32 const&) const':
/dnp3/cpp/libs/opendnp3/src/opendnp3/app/AnalogOutput.cpp:55: multiple definition of `opendnp3::AnalogOutputInt32::operator==(opendnp3::AnalogOutputInt32 const&) const'
cpp/libs/opendnp3/src/opendnp3/app/.libs/libopendnp3_la-AnalogOutput.o:/dnp3/cpp/libs/opendnp3/src/opendnp3/app/AnalogOutput.cpp:55: first defined here

````

jadamcrain, 2015-07-19:
Is this the tip of 2.0.x or  compiled release?

Is this the official build or are you creating your own program or just building the included demos?

Also, please include your toolchain versions

tweakmy, 2015-07-20:
$ ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for style of include used by make... GNU
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking whether g++ supports C++11 features by default... no
checking whether g++ supports C++11 features with -std=c++11... yes
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking how to print strings... printf
checking for gcc... gcc
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
configure: --with-lcov=no
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: executing depfiles commands
config.status: executing libtool commands

tweakmy, 2015-07-20:
Just building the vanila version, no extra on in on branch 2.0.x.

tweakmy, 2015-07-20:
build works on the tags 2.0.1-RC3

jadamcrain, 2015-07-20:
But fails on 2.0.1 final?

closed 2015-08-23 by jadamcrain