dnp3/opendnp3 #140: Make ChannelSet a public interface

atmurray, 2016-01-13:
Can asiodnp3/ChannelSet.h be moved so that it is a public header?
This will enable alternates to DNP3Manager to be created outside the library.

jadamcrain, 2016-01-16:
It's certainly easy to do so, but isn't this a slippery slope? How about an option in CMake to install all headers?

I'd also be curious to hear about the use case that requires an alternative to the current API. I have a purchase order coming for a GPRS-type master API where the master accepts connections and creates dynamic sessions.

atmurray, 2016-01-17:
I did wonder if it was a slippery slope. The two drivers I have are
 - finer control over logging (such that loggers can subscribe to particular channels, masters and outstations); and
 - re-use of an existing asio::io_service
I guess both of these things could instead be achieved through extensions to DNP3Manager?

jadamcrain, 2016-01-18:
Yes, both of these are quite achievable. I would rather see API generalizations, than people having to build custom code.

1) We can make the "LogRoot" for the session/channel distinct, and make both adjustable at run-time. The API for IMaster/IOutstation could have runtime way of changing log level separate from the channel itself.

2) We can add an overloaded constructor to DNP3Manager that takes an io_service.

WRT 2), we could even generalize how the IOService gets "run".

jadamcrain, 2016-11-14:
ChannelSet no longers exists. Most of the startup/teardown machinery has been simplified in 2.2.0-RC1

closed 2016-11-14 by jadamcrain