dnp3/opendnp3 #142: Command fail

alex529, 2016-01-28:
When an ICommandCollection colection is created from a CommandSet and nothing is added to it the DirectOperate comand will fail with the fallwing message:
 
info: Summary: FAILURE_INTERNAL_ERROR
 Header: 1  Index: 3  State: INIT Status: UNDEFINED

```
opendnp3::CommandSet commands;
auto& aod64 = commands.StartHeader<opendnp3::AnalogOutputDouble64>();// this will make the command fail
auto& crb = commands.StartHeader<opendnp3::ControlRelayOutputBlock>();
opendnp3::ControlRelayOutputBlock crob{opendnp3::ControlCode::LATCH_OFF};
crobs.push_back(std::make_pair(crob, dp.index));
master->DirectOperate(std::move(commands), callback);
```

I'm not sure if this is buy design or it's a bug but its not documented anywh

jadamcrain, 2016-01-28:
Sounds like we should just ignore any empty headers in the CommandSet. The outstation will explicitly refuse to parse requests with a header count of zero.

Will investigate before next release candidate.

jadamcrain, 2016-01-29:
What's going on here is the task is failing to format the request, so it returns "FAILURE_INTERNAL_ERROR".  There was already a safeguard to check for an empty header definition:

https://github.com/automatak/dnp3/blob/2.0.x/cpp/libs/src/opendnp3/master/TypedCommandHeader.h#L103

I just added a test case that demonstrates that the master immediately fails the task and doesn't transmit this nonsensical request:

https://github.com/automatak/dnp3/commit/c321070ffe48693264e0608a2136578fa3cd3f11

Not sure what to do to make this more clear code wise, but we could mention this in the docs or FAQ.

closed 2016-03-01 by jadamcrain