dnp3/opendnp3 #146: ISOEHandler's HeaderInfo class lacks easy way to tell static from event data

jadamcrain, 2016-02-11:
Recommend adding a field or enum to tell the user the type of data w/o making them know all the groups/variations.

sidhoda, 2016-02-15:
This would be useful together with a field/enum to state if the event has a timestamp. We currently use the following C# code for this, which might be a good starting point:

```csharp
public static class DnpGroupVariationHelper
{
	public static bool IsEventVariation(this GroupVariation variation)
	{
		switch (variation)
		{
			// binary input
			case GroupVariation.Group2Var0:
			case GroupVariation.Group2Var1:
			case GroupVariation.Group2Var2:
			case GroupVariation.Group2Var3:
			// binary output
			case GroupVariation.Group11Var0:
			case GroupVariation.Group11Var1:
			case GroupVariation.Group11Var2:
			// binary output command
			case GroupVariation.Group13Var1:
			case GroupVariation.Group13Var2:
			// counter input
			case GroupVariation.Group22Var0:
			case GroupVariation.Group22Var1:
			case GroupVariation.Group22Var2:
			case GroupVariation.Group22Var5:
			case GroupVariation.Group22Var6:
			// analogue input
			case GroupVariation.Group32Var0:
			case GroupVariation.Group32Var1:
			case GroupVariation.Group32Var2:
			case GroupVariation.Group32Var3:
			case GroupVariation.Group32Var4:
			case GroupVariation.Group32Var5:
			case GroupVariation.Group32Var6:
			case GroupVariation.Group32Var7:
			case GroupVariation.Group32Var8:
			// analogue output
			case GroupVariation.Group42Var0:
			case GroupVariation.Group42Var1:
			case GroupVariation.Group42Var2:
			case GroupVariation.Group42Var3:
			case GroupVariation.Group42Var4:
			case GroupVariation.Group42Var5:
			case GroupVariation.Group42Var6:
			case GroupVariation.Group42Var7:
			case GroupVariation.Group42Var8:
			// analogue output command
			case GroupVariation.Group43Var1:
			case GroupVariation.Group43Var2:
			case GroupVariation.Group43Var3:
			case GroupVariation.Group43Var4:
			case GroupVariation.Group43Var5:
			case GroupVariation.Group43Var6:
			case GroupVariation.Group43Var7:
			case GroupVariation.Group43Var8:
				return true;
			default:
				return false;
		}
	}

	public static bool HasTimestamp(this GroupVariation variation)
	{
		switch (variation)
		{
			// binary input event
			case GroupVariation.Group2Var2:
			case GroupVariation.Group2Var3:
			// double-bit binary input
			case GroupVariation.Group4Var2:
			case GroupVariation.Group4Var3:
			// binary output event
			case GroupVariation.Group11Var2:
			// binary output command event
			case GroupVariation.Group13Var2:
			// frozen counter status
			case GroupVariation.Group21Var5:
			case GroupVariation.Group21Var6:
			// counter input event
			case GroupVariation.Group22Var5:
			case GroupVariation.Group22Var6:
			// frozen counter event
			case GroupVariation.Group23Var5:
			case GroupVariation.Group23Var6:
			// analogue input event
			case GroupVariation.Group32Var3:
			case GroupVariation.Group32Var4:
			case GroupVariation.Group32Var7:
			case GroupVariation.Group32Var8:
			// analogue output event
			case GroupVariation.Group42Var3:
			case GroupVariation.Group42Var4:
			case GroupVariation.Group42Var7:
			case GroupVariation.Group42Var8:
			// analogue output command event
			case GroupVariation.Group43Var3:
			case GroupVariation.Group43Var4:
			case GroupVariation.Group43Var7:
			case GroupVariation.Group43Var8:
				return true;
			default:
				return false;
		}
	}
}
```

jadamcrain, 2016-02-15:
There are already fields in HeaderInfo that tell you about the validity of the timestamp and how the qualify was derived:

https://github.com/automatak/dnp3/blob/2.0.x/dotnet/bindings/CLRInterface/ISOEHandler.cs#L76

I'll try and implement a hint for event/static, but do it using the code generator to lessen the chance of future misimplementation.

jadamcrain, 2016-02-15:
Looks like the thing about the flags wasn't carried through from C++:

https://github.com/automatak/dnp3/blob/2.0.x/cpp/libs/include/opendnp3/master/HeaderInfo.h#L68

jadamcrain, 2016-03-18:
I added a field to HeaderInfo that tells if the header is for an event type. BTW @sidhoda,  I'd switch to this because the switches above are missing some types.

I also added the "flagsValid" field to the C# version.

closed 2016-03-18 by jadamcrain