dnp3/opendnp3 #147: Keep alive timer is not reset on receiving initial null unsolicited message

sidhoda, 2016-03-01:
If while waiting for a link status response, the outstation powers up and sends the initial null unsolicited message then the keep alive timer is not reset and will fail and result in another link status request being sent.

This results in the following sequence of events:

- keep alive initiated
- outstation powers up and null unsolicited received (outstation did not receive the link status request)
- keep alive failed
- keep alive initiated
- keep alive success
- proceed with startup tasks

The DNP3 spec is a little unclear on what should happen here, and it is arguably correct as the outstation didn't response to the link status request even though it send some data is clearly communicating again. This behaviour doesn't cause too much problem as the link is eventually re-established, however I thought it would be worth raising and discussing if a change is required.

jadamcrain, 2016-03-03:
I think that each side should definitely wait for a response, but what's more puzzling to me is that the master shouldn't be sending a REQUEST_LINK_STATES right after the session begins. It should have to wait for a some configured period with no traffic before it does this. Is it sending it right after the connection is established?

Secondly, the outstation should definitely not delay in sending LINK_STATUS response even if is waiting for a confirm at the application layer.

jadamcrain, 2016-10-23:
@sidhoda Is this bug still valid?

sidhoda, 2016-11-19:
I'll need to retest this to remind myself what was happening, but I *think* the master sends the link status request on activity timeout, N seconds after null unsolicited is received. Once the outstation receives it, it does respond immediately. Not actually sure this is a bug though, and it's not something that's been noticed in production

closed 2017-12-16 by jadamcrain