dnp3/opendnp3 #150: compile errors

luzik, 2016-03-09:
While using my crosscompile compilator i found this issue:

```
[ 67%] Building CXX object CMakeFiles/dnp3decode.dir/cpp/libs/src/dnp3decode/LoggingHandler.cpp.o
/opt/poky/1.3/sysroots/x86_64-pokysdk-linux/usr/bin/armv5te-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++   -march=armv5te  -marm -mthumb-interwork  -mtune=arm926ej-s --sysroot=/opt/poky/1.3/sysroots/armv5te-poky-linux-gnueabi  -DASIO_STANDALONE -DOPENDNP3_USE_SECAUTH -DOPENDNP3_USE_TLS -Ddnp3decode_EXPORTS -Wall -std=c++11 -fPIC -I/home/luzik/dnp3/asio/asio/include -I/home/luzik/dnp3/./cpp/libs/src -I/home/luzik/dnp3/./cpp/libs/include -I/home/luzik/dnp3/./cpp/tests/libs/src    -o CMakeFiles/dnp3decode.dir/cpp/libs/src/dnp3decode/LoggingHandler.cpp.o -c /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:0:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h: In lambda function:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:164:4: error: no matching function for call to 'openpal::Logger::Log(const int32_t&, const char [63], char [120], int) const'
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:164:4: note: candidate is:
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:24:0,
                 from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note: void openpal::Logger::Log(const openpal::LogFilters&, const char*, const char*, int) <near match>
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note:   no known conversion for implicit 'this' parameter from 'const openpal::Logger*' to 'openpal::Logger*'
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:0:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h: In lambda function:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:180:4: error: no matching function for call to 'openpal::Logger::Log(const int32_t&, const char [63], char [120], int) const'
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:180:4: note: candidate is:
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:24:0,
                 from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note: void openpal::Logger::Log(const openpal::LogFilters&, const char*, const char*, int) <near match>
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note:   no known conversion for implicit 'this' parameter from 'const openpal::Logger*' to 'openpal::Logger*'
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:0:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h: In lambda function:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:198:4: error: no matching function for call to 'openpal::Logger::Log(const int32_t&, const char [63], const char*, int) const'
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:198:4: note: candidate is:
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:24:0,
                 from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note: void openpal::Logger::Log(const openpal::LogFilters&, const char*, const char*, int) <near match>
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note:   no known conversion for implicit 'this' parameter from 'const openpal::Logger*' to 'openpal::Logger*'
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:0:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h: In lambda function:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:223:4: error: no matching function for call to 'openpal::Logger::Log(const int32_t&, const char [63], const char*, int) const'
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:223:4: note: candidate is:
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:24:0,
                 from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note: void openpal::Logger::Log(const openpal::LogFilters&, const char*, const char*, int) <near match>
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note:   no known conversion for implicit 'this' parameter from 'const openpal::Logger*' to 'openpal::Logger*'
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:0:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h: In lambda function:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:241:4: error: no matching function for call to 'openpal::Logger::Log(const int32_t&, const char [63], const char*, int) const'
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:241:4: note: candidate is:
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:24:0,
                 from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note: void openpal::Logger::Log(const openpal::LogFilters&, const char*, const char*, int) <near match>
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note:   no known conversion for implicit 'this' parameter from 'const openpal::Logger*' to 'openpal::Logger*'
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:0:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h: In lambda function:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:258:4: error: no matching function for call to 'openpal::Logger::Log(const int32_t&, const char [63], const char*, int) const'
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:258:4: note: candidate is:
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:24:0,
                 from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note: void openpal::Logger::Log(const openpal::LogFilters&, const char*, const char*, int) <near match>
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note:   no known conversion for implicit 'this' parameter from 'const openpal::Logger*' to 'openpal::Logger*'
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:0:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h: In lambda function:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:283:4: error: no matching function for call to 'openpal::Logger::Log(const int32_t&, const char [63], const char*, int) const'
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:283:4: note: candidate is:
In file included from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.h:24:0,
                 from /home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:22:
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note: void openpal::Logger::Log(const openpal::LogFilters&, const char*, const char*, int) <near match>
/home/luzik/dnp3/./cpp/libs/include/openpal/logging/Logger.h:43:7: note:   no known conversion for implicit 'this' parameter from 'const openpal::Logger*' to 'openpal::Logger*'
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp: In lambda function:
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:407:78: error: 'this' was not captured for this lambda function
/home/luzik/dnp3/cpp/libs/src/dnp3decode/LoggingHandler.cpp:407:81: warning: control reaches end of non-void function [-Wreturn-type]
make[2]: *** [CMakeFiles/dnp3decode.dir/cpp/libs/src/dnp3decode/LoggingHandler.cpp.o] Błąd 1

```

Other ones are fixed by:

[... 15 lines cut ...]
--- a/cpp/libs/src/opendnp3/master/MeasurementHandler.h
+++ b/cpp/libs/src/opendnp3/master/MeasurementHandler.h
@@ -101,7 +101,7 @@ private:
        template <class Target, class Source>
        IINField LoadValuesWithTransformTo(const HeaderRecord& record, const ICollection<Indexed<Source>>& values)
        {
-               auto transform = [](const Indexed<Source>& input) -> Indexed<Target>
+               auto transform = [&](const Indexed<Source>& input) -> Indexed<Target>
                {
[... cut ...]  Target>(input);
                };
diff --git a/cpp/libs/src/secauth/outstation/IOutstationUserDatabase.h b/cpp/libs/src/secauth/outstation/IOutstationUserDatabase.h
index 120b1bf..5d6c127 100644
--- a/cpp/libs/src/secauth/outstation/IOutstationUserDatabase.h
+++ b/cpp/libs/src/secauth/outstation/IOutstationUserDatabase.h
@@ -28,6 +28,7 @@
 #include <openpal/container/RSlice.h>
 
 #include <functional>
+#include <string>
 
 #include "secauth/UpdateKey.h"
```

Please help me deal with first one.

jadamcrain, 2016-03-09:
compiler and version?

luzik, 2016-03-10:
to old ..excuse me for ignoring requirements informations

closed 2016-03-10 by luzik