dnp3/opendnp3 #151: Issue with multiple sessions sharing same connection

spacecoder007, 2016-03-11:
When there are 2 outstations on the same connection with different source addresses, the master cannot poll the outstation it is connected to. It seems to hang.

Issue can be reproduced using the simulator:

Channel with 2 masters, both on the same connection 127.0.0.1:20000
master: SRC 1 DEST 2
master2: SRC 1 DEST 3

Channel with 2 outstations, both listening on 127.0.0.1:20000
outstation: SRC 2 DEST 1
outstation2: SRC 3 DEST 1

master connects and polls outstation OK
master2 connects and polls outstation2 OK

Close the master, then on polling outstation2 from master2, no response and nothing is logged.

I've attached the log files.

[masters.txt](https://github.com/automatak/dnp3/files/169357/masters.txt)
[outstations.txt](https://github.com/automatak/dnp3/files/169358/outstations.txt)

jadamcrain, 2016-03-11:
Thanks Darren. It may be a little while before I get a chance to take a look into this.

sidhoda, 2016-03-11:
Just to highlight the relevant part of the log file. When polling outstation2 from master2, the following line is logged:

14:27:13.925  INFO    channel       Begining task: Application Poll

No bytes are sent across the network for the poll (confirmed using Wireshark) so it appears that the poll is just ignored. Also, the Task returned from calling IMaster.Scan never completes.

If I find time, I'll try to search down why this is happening in the code.

sidhoda, 2016-03-17:
I think I can explain what is happening... when calling Shutdown on the first master it calls OnLayerDown on the shared MultiDropTaskLock, 

This then sets the task lock into an offline state, even though the second master is still online. 

When the next poll comes in it can't acquire a lock so the task never completes.

I guess the fix would be to only call pTaskLock->OnLayerDown if it is not being used by another master stack?

jadamcrain, 2016-03-17:
This is probably enough for me to go on to fix it. The way multiple masters share a session currently is pretty awkward.

sidhoda, 2016-03-17:
We have one system that's bringing back field telemetry via a third-party data concentrator over a single TCP connection.

It is a rare requirement, however we are currently blocked from deploying opendnp3 onto that system due to this issue.

sidhoda, 2016-03-18:
Would a simple fix for this to add a reference counter to the MultiDropTaskLock, which is incremented on OnLayerUp and decremented on OnLayerDown. When getting to zero, it would do the usual cleanup?

jadamcrain, 2016-03-18:
Quite possibly, I'm having a look now. I'm hoping I can replicate this with an integration test instead of testing manually.

jadamcrain, 2016-03-18:
Looking at the MultidropTaskLock, this is quite clearly broken. the use case of removing 1 master from an active set of N sessions is not handled.

I am going to refactor this, write some new tests, and then test the behavior manually too using the simulator.

jadamcrain, 2016-03-18:
In e125f8455cf4a43ee5dd2ffdc37c I added a test that for this behavior.  In the commit before it I've altered the ITaskLock behavior so there is no concept of being "offline". I don't think this extra behavior is really needed.

Can you try the "fix-task-lock" branch with the simulator and see if this resolves the integrated behavior?

-Adam

jadamcrain, 2016-03-18:
In 09a92ac8228034d5eadcf44306d6745e3b503023 I changed ownership of the MultidropTaskLock to the LinkLayerRouter and added methods to set it online/offline there. This prevents callbacks to start tasks when channel is closing.

I think this matches the original intent of what I was trying to accomplish, but puts the call in the correct place.

spacecoder007, 2016-03-18:
Hi Adam, just updated the simulator to use the latest from fix-task-lock branch and retested the scenario above and still getting the same issue.

spacecoder007, 2016-03-18:
Will try it again with the change you just committed.

jadamcrain, 2016-03-18:
How are you updating the simulator.. by replacing it with a fully built local Nuget package?

spacecoder007, 2016-03-18:
I build the fix-task-branch and put those DLLs into the packages folder for the simulator to use when building in VS. Then I fixed errors due to simulator being out of date. I just tested it again and it worked! Thanks for fixing it. Attached the log files to show it worked.

[masters-fixed.txt](https://github.com/automatak/dnp3/files/179976/masters-fixed.txt)
[outstations-fixed.txt](https://github.com/automatak/dnp3/files/179975/outstations-fixed.txt)

jadamcrain, 2016-03-18:
Excellent! I will merge this branch into 2.0.x now then. I will also merge 2.0.x into the 'gprs-mem-fix' branch so that you have both together while testing the new master API.

spacecoder007, 2016-03-18:
Thanks. Is the gprs branch redundant? Is all the changes from that branch in gprs-mem-fix?

jadamcrain, 2016-03-18:
Yes. The gprs-mem-fix branch also contains the TLS work. Let me do all this merging and I will send you and Sid a private email.

closed 2016-03-18 by jadamcrain