dnp3/opendnp3 #154: Add support for disabling link status requests

spacecoder007, 2016-03-24:
Currently if keep alive timeout is not specified it uses a default. Would it be possible to disable keep alive timer by possibly setting another flag on LinkConfig?

jadamcrain, 2016-03-24:
In C++, the underlying representation for TimeDuration is int64_t, so you can effectively set it so large it will never happen. In C#, for some reason, I chose not to use TimeSpan and  used 32-bit milliseconds, so it will actually run the timer after 49 days :/.

If I switch this to be a TimeSpan that you can effectively set to infinity, is this acceptable?

spacecoder007, 2016-03-24:
Yep that sounds a good solution. Thanks.

jadamcrain, 2016-03-24:
Ok, let me know how this works and I'll merge this:

https://github.com/automatak/dnp3/blob/link-config-timespan-fix/dotnet/bindings/CLRInterface/config/LinkConfig.cs#L160

spacecoder007, 2016-03-24:
Just tested it and setting it to TimeSpan.MaxValue worked fine.

spacecoder007, 2016-03-24:
Thanks for changing it :)

jadamcrain, 2016-03-25:
No problem. Anytime you notice anything inconsistent about the API, please let me know even if it's not a usage issue.

I've merged this into 2.0.x

closed 2016-03-25 by jadamcrain