dnp3/opendnp3 #164: Best way to process repeat read requests

bmironenko, 2016-05-19:
In our application the master (which is not under our control) may sometimes send repeat read requests (same sequence ID), which must not be ignored by our outstation.

I'd like to propose a pull request that would add the boolean `OutstationParams.ignoreRepeatReads` attribute. It would get initialized to `true` by default, but when `false`, it would alter the logic in `OContext::OnReceiveSolRequest` to respond to the request as if it is new (i.e. call and return `OContext::ProcessNewRequest`). This seems to work for us, but I'm curious if there is a better/preferred way from the architectural point of view.

jadamcrain, 2016-05-23:
What does the spec have to say on how an outstation should handle repeat READ requests with the same sequence #? 

In general, I am not in favor of adding work-around flags for non-conformant implementations unless there  absolutely no alternative. Is there any way to force (or perhaps more realistically "encourage") the entity responsible for the master to patch their system by pointing out that it's non-conformant?

blytkerchan, 2016-05-23:
Section 4.3 rule 14:

   -

   For each new, non-retry request fragment it sends, a master shall
   increment the SEQ number by one count (modulo 16) from its previous request
   fragment.

The master is clearly not conformant in this case.

OTOH: there is no harm in the Outstation responding to a properly formatted
request, regardless of the sequence number of that request. The way I see
it, the sequence numbers are mostly there to help the master keep track of
which response goes with which request: they have very little use for the
Outstation.

rlc

bmironenko, 2016-05-23:
To clarify: master issues a read request every 2 seconds. Sample sequence of events:

1. Master sends a read request
2. Outstation responds but delivery takes longer than 2 seconds due to network congestion
3. Master times out after 2 seconds and sends a repeat read request
4. Master receives the original response but throws it out even though it contains the sequence ID that has already been sent twice
5. Master times out again (since the outstation ignored the repeat read request)

I'm not sure which layer on the master side is responsible for (4), but common sense says that the response should not be thrown out since it matches the sequence number of an outstanding (repeat) request.

Unfortunately, it is unlikely that the master entity would be willing to update their implementation, especially given the workaround we've been using.

jadamcrain, 2016-05-24:
As far as I can tell, opendnp3 never "ignores" repeat read requests. As per the outstation state tables on pg 157, if a repeat request is received w/ the same sequence # and is an octet-for-octet match of a previous request, it should re-transmit the same response.

This is really designed to prevent "double operations". One could imagine that executing a pulse-train via a CROB twice could be undesirable.

I'll have to look at the unit test, but I'd be very surprised if opendnp3 was just ignoring repeat reads.

jadamcrain, 2016-10-17:
Confirmed that this is indeed a bug. I've put a patch onto the tip of 2.0.x, but plan to do a full audit of the outstation state machine before the next release. I'll leave this ticket open until that time.

jadamcrain, 2016-10-20:
I refactored the outstation state machine to line up w/ 1815-2012. This was originally written for an older version of the standard that had separate state machines for solicited/unsolicited. Much cleaner now and this issue is resolved on the "development branch". Hoping to have 2.2.0-RC1 out sometime in the next 2 weeks.

closed 2016-10-20 by jadamcrain