dnp3/opendnp3 #167: Issues with setting Event Buffer to Maximum Value for an Outstation

spacecoder007, 2016-06-29:
I have 1000 outstations, each with 16 analogs. The simulator aborts when setting either of the following:

config.outstation.buffer.maxAnalogEvents = uint.MaxValue;

OR

config.outstation.buffer = new EventBufferConfig(uint.MaxValue);

It seems Automatak.DNP3.Interface.EventBufferConfig uses uint, however the underlying code in opendnp3/outstation/EventBufferConfig.cpp uses uint16_t, so one fix is to change EventBufferConfig to use ushort.

It will work when using byte.MaxValue. Maybe there is an overflow issue here?

jadamcrain, 2016-07-05:
Thanks Darren, will have a look before next release. It's probably an exception being thrown in a Convert.ToUInt16 method. Best solution is probably to change the .NET type to match the C++ type so that an exception is impossible.

jadamcrain, 2016-07-06:
I tested this with the debugger and the C# outstation example. UInt32.MaxValue was truncated without an exception to 65535... odd that compiler gives no warnings... I will look into the settings to see if this can be explicitly tweaked.

Any idea how/why the simulator aborts?

I've committed a fix to make the C# config type be UInt16.

spacecoder007, 2016-07-11:
Hi Adam,

To reproduce this issue, I am doing the following:

```
IDNP3Manager mgr = DNP3ManagerFactory.CreateManager(new PrintingLogAdapter());
for (ushort i = 20000; i < 21000; i++)
{
    var channel = mgr.AddTCPServer("server" + i, 
        LogLevels.NONE, ChannelRetry.Default, "0.0.0.0", i);
    var config = new OutstationStackConfig();
    config.link.isMaster = false;
    config.link.localAddr = 2;
    config.link.remoteAddr = 1;
    config.outstation.buffer = new EventBufferConfig(ushort.MaxValue);
    var outstation = channel.AddOutstation("outstation" + i, 
        RejectingCommandHandler.Instance, DefaultOutstationApplication.Instance, config);
    outstation.Enable();
}
```

No master is running. On reaching port 20066 it aborts.

[image]

This happens when using uint.MaxValue and ushort.MaxValue. It works fine when using byte.Maxvalue.

I haven't been able to find the cause as of yet.

jadamcrain, 2016-07-29:
I enabled native debugging:

"First-chance exception at 0x76a8dae8 (KernelBase.dll) in DNP3CLROutstationDemo.exe: Microsoft C++ exception: std::bad_alloc at memory location 0x06bff538."

native C++ exception b/c the operating system failed to give the program memory.

This is not really unexpected if you're allocating 65535 event buffer slots for every type.

sizeof(openpal::ListNode<opendnp3::SOERecord>) == 56 bytes.

65535*56*8 data types = 29359680 == ~30MB

So, just the event buffers alone, ignoring all other dynamic memory allocated for other buffers, are 30MB per instance.

30MB*66 = 1980 = ~2GB.

-Adam

sidhoda, 2016-08-02:
Makes perfect sense, especially when we were trying to allocate uint.MaxValue per buffer :-S

The detail we were missing was buffers are pre-allocated rather than dynamically growing up to the set maximum.

Go ahead and close this.

jadamcrain, 2016-08-03:
I'm actually going to reopen this as there is room for improvement in the documentation here. We should document that buffers are preallocated in the EventBufferConfig class.