dnp3/opendnp3 #168: Channel state listener causes the .NET CLR to crash when running as part of a unit test

tspeckert, 2016-07-11:
While trying to create some automated integration tests, I encountered an interesting issue which causes the CLR to crash when I attempt to run an opendnp3 outstation inside of a "unit test" (I've tried NUnit 2.6.4, 3.4.0, and Visual Studio's unit test framework). If I add a channel state listener to a channel, the .NET CLR will crash (FatalExecutionEngineError) shortly after adding it (not immediately, there seems to be a short delay). This happens no matter which test runner I use. The following log appears in the event viewer:

Application: JetBrains.ReSharper.TaskRunner.CLR45.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an internal error in the .NET Runtime at IP 71518F26 (71500000) with exit code 80131506.

If I comment out the line where the channel state listener is added, then everything works as expected. If I run the code in a "normal" application, (i.e. create a console application and run the test method from there), then everything runs properly. The error only seems to occur when running the code within a unit test.

Here is a minimal example which reproduces the problem with Nunit 2.6.4:

```csharp
using System;
using System.Threading;
using Automatak.DNP3.Adapter;
using Automatak.DNP3.Interface;
using NUnit.Framework;

namespace DNP3TestProblem
{
    [TestFixture]
    public class Tests
    {
        [Test]
        public void BreakingTest()
        {
            IDNP3Manager mgr = DNP3ManagerFactory.CreateManager(1, PrintingLogAdapter.Instance);
            IChannel channel = mgr.AddTCPServer("server", LogLevels.NORMAL, ChannelRetry.Default, "127.0.0.1", 20000);
            channel.AddStateListener(state => Console.WriteLine("channel state: " + state));

            Thread.Sleep(20);
            Console.WriteLine(20);

            Thread.Sleep(50); 
            Console.WriteLine(50); 

            Thread.Sleep(50);
            Console.WriteLine(50);

            Thread.Sleep(100);
            Console.WriteLine(100);

            Thread.Sleep(1000);
            Console.WriteLine(1000);
        }
    }
}
```

jadamcrain, 2016-10-05:
Thomas,

I've changed the way most user supplied callback interfaces are injected in C++, and I believe this will resolve the issue.

Can you try the development branch out with your unit test setup when possible?

thanks,
Adam

tspeckert, 2016-10-24:
Hi Adam,

Sorry for the slow response, I was away for some time. Unfortunately the issue still seems to be occurring within both NUnit and MS unit testing projects (tested against the development branch, revision 2d70e3b).

Here's my modified code:
```
using System;
using System.Threading;
using Automatak.DNP3.Adapter;
using Automatak.DNP3.Interface;
using NUnit.Framework;

namespace DNP3TestProblem
{
    [TestFixture]
    public class Tests
    {
        [Test]
        public void BreakingTest()
        {
            IDNP3Manager mgr = DNP3ManagerFactory.CreateManager(1, new PrintingLogAdapter());
            IChannel channel = mgr.AddTCPServer("server", LogLevels.NORMAL, ChannelRetry.Default, "127.0.0.1", 20000, new MyListener());

            Thread.Sleep(20);
            Console.WriteLine(20);

            Thread.Sleep(50);
            Console.WriteLine(50);

            Thread.Sleep(50);
            Console.WriteLine(50);

            Thread.Sleep(100);
            Console.WriteLine(100);

            Thread.Sleep(1000);
            Console.WriteLine(1000);
        }

    }

    public class MyListener : IChannelListener
    {
        public void OnStateChange(ChannelState state)
        {
            Console.WriteLine("channel state: " + state);
        }
    }
}
```

jadamcrain, 2016-10-24:
Thanks. I will set this up myself and use the debugger to figure out what's going on.

Having a test suite of "smoke tests" for the .NET would be a good thing anyway, so I'll likely leave these tests there permanently.

-Adam

jadamcrain, 2016-10-24:
I've added a unit test suite w/ this test.  It starts to run, but then fails with this weird message:

------------
System.IO.FileNotFoundException: Could not load file or assembly 'DNP3CLRInterface, Version=2.0.1.1, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
File name: 'DNP3CLRInterface, Version=2.0.1.1, Culture=neutral, PublicKeyToken=null'
   at Automatak.DNP3.Adapter.LogAdapter.Log(LogAdapter* , LogEntry* Entry)
-----------

The same happened with NUnit. This is really weird, b/c it happens before the test even finishes, and this really isn't any different than the example programs.

sidhoda, 2016-10-24:
Could it be that the NUnit runner isn't running as x86?

Fusion Log Viewer might help with working out where the assembly binding is failing.

jadamcrain, 2016-10-24:
It happens with the NUnit Runner plugin for visual studio, but also with the built-in Visual Studio test runner.

I've definitely set the test DLL to be Win32/64 based on build type.

Can you try this on your machine also @sidhoda ? It would be nice to have a simple regression test suite for .NET

tspeckert, 2016-10-26:
With nunit there's a separate executable for 32-bit and 64-bit test runners. On Nunit3 you can just give an option to nunit3-console so it uses the x86 version, but on Nunit 2.6.4, it's necessary to run nunit3-console-x86. With the visual studio plugins, it *should* pick the correct one based on the selected platform I believe...

I'll build/run a debug version later on today.

jadamcrain, 2016-10-26:
Thanks Thomas. So more info:

I don't believe this is the same issue you were experiencing. The fault happens regardless of whether the channel is created or not sometime after the manager is constructed. It appears to happen when the first log callback happens when a thread pool thread is first run.

tspeckert, 2016-10-27:
Alright, so the 'FileNotFoundException' seems to occur when the test runner runs under an app domain that's different than the test itself. To fix that I added the argument '--domain=NONE' to nunit3-console.exe (I had to also copy nunit.framework.dll to the same directory as the test runner... though I don't exactly know why).

So back to the original problem: If run nunit from the console (not within VS), my test build (commit cff2fef, built using the DEBUG option in cmake and VS) successfully runs the test now. For some reason if I run it from within visual studio, the test runner crashes and this appears in the event log:
'''
Application: JetBrains.ReSharper.TaskRunner.CLR4.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.ArgumentException
   at System.Runtime.InteropServices.GCHandle.InternalCheckDomain(IntPtr)
   at System.Runtime.InteropServices.GCHandle.FromIntPtr(IntPtr)
   at System.Runtime.InteropServices.GCHandle.op_Explicit(IntPtr)
   at <Module>.gcroot<Automatak::DNP3::Interface::ILogHandler ^>.->(gcroot<Automatak::DNP3::Interface::ILogHandler ^>*)
   at <Module>.Automatak.DNP3.Adapter.LogAdapter.Log(Automatak.DNP3.Adapter.LogAdapter*, openpal.LogEntry*)
'''

I suspect this is what I saw when I ran my updated test, and I mistook it for the original error. Also, if LogAdapter.Log() is called on a background thread, then nunit wouldn't actually see the exception (but there could be something in the VS plugins that is detecting it and aborting the tests).

sidhoda, 2016-11-01:
Tested this out on my machine and reproduced the FileNotFoundException using both VS test runner and mstest.exe.

From the assembly binding logs, I can see it is trying to probe for the DLLs in C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\IDE\ which is where the test execution engine (QTAgent32_40.exe) is running from. 

Copying both DNP3CLRInterface.DLL and DNP3CLRAdapter.DLL into this folder results in the test passing.

Also, running mstest.exe with the /noisolation option results in the test passing.

jadamcrain, 2016-11-02:
Ok, so it sounds to me like the error is just a defect in the way the test runners handle mixed-mode assemblies. 

Until we have the time to write some proper integration tests, I'm inclined to delete the test and close this issue as it was originally raised b/c of a defect in the bindings that appears to now be resolved.

Comments?

If someone has time during the RC process to add a good integration test we can certainly do it, but I'm really stretched thin and don't want to delay the RC for this.

sidhoda, 2016-11-03:
No comments from me, sounds like a good way forward.

tspeckert, 2016-11-03:
Also sounds good to me. One comment though: the System.ArgumentException I posed above comes from the logger- is there another issue somewhere where LogAdapter.Log is being caleld with invalid arguments?

sidhoda, 2016-11-03:
I didn't see that exception when running the unit test with no isolation. Does your unit test match the one on the development branch?

tspeckert, 2016-11-08:
I believe it did (although nunit was used), but I appear to have deleted the exact code. In any case, I just updated to the latest development commit and re-ran the test from the development branch: it passes with no issues. I also tried them under nunit (all good) and with the following addition to make sure nothing was happening in the background:

```
static ManagerSmokeTests()
{
    AppDomain.CurrentDomain.UnhandledException += (sender, args) =>
    {
        Console.WriteLine($"unhandled exception caught: {args.ExceptionObject}");
        Assert.Fail("unhandled exception occured.");
    };
}
```

Looks good to me.

closed 2016-11-08 by tspeckert