dnp3/opendnp3 #173: TLSClient(): connection unrecoverable after comm glitch

vanjabucic, 2016-09-01:
(carrying over my question on google groups, making an issue out of it)
(https://groups.google.com/forum/#!topic/automatak-dnp3/Zchb4aMyKhM)

I have been trying to use TLS protected communications, and while it is successful at first it soon enters unrecoverable state where it is either OPENING or WAITING. Looks like errors in the TLS layer are not triggering a full connection reset - instead it is trying to limp along with bad and disconnected link. 

[log fragment] 
... 
good connection... 
... 
18:54:53.318 dc outstation.log  ms(1472597689553)                         WARN      server[AC1] - decryption failed or bad record mac   
18:54:53.318 dc outstation.log  ms(1472597689553)                         WARN      server[AC1] - Operation aborted.   
18:54:53.318 dc outstation.log  [2016-08-30 18:54:49.553928] [0xed89f700] [warning] server[AC1] state: WAITING   
18:54:53.319 dc outstation.log  [2016-08-30 18:54:50.554318] [0xed89f700] [warning] server[AC1] state: OPENING   
18:54:53.319 dc outstation.log  ms(1472597690566)                         WARN      server[AC1] - error: excessive message size   
18:54:53.319 dc outstation.log  [2016-08-30 18:54:50.567162] [0xed89f700] [warning] server[AC1] state: WAITING   
18:54:53.319 dc outstation.log  [2016-08-30 18:54:51.567550] [0xed89f700] [warning] server[AC1] state: OPENING   
18:54:53.319 dc outstation.log  ms(1472597691578)                         WARN      server[AC1] - error: excessive message size 
... 
18:54:58.336 dc outstation.log  [2016-08-30 18:54:50.567162] [0xed89f700] [warning] server[AC1] state: WAITING   
18:54:58.336 dc outstation.log  [2016-08-30 18:54:51.567550] [0xed89f700] [warning] server[AC1] state: OPENING   
18:54:58.336 dc outstation.log  ms(1472597691578)                         WARN      server[AC1] - error: excessive message size 
... 
09:41:11.543 dc outstation.log  ms(1472737268856)                         WARN    server[AC2] - error: short read 
... 
[log fragment] 

Is this a bug, or something user code needs to deal with?

If the underlying library code will not reset/refresh the connection, how can the user code detect this and force a reconnect?

Thanks,
Vanja

jadamcrain, 2016-09-01:
Thanks for the report.

This is either a bug in opendnp3, or some cross TLS implementation/version problem. The user should not have intervene to reset the issue.

Can you update this ticket with the following additional info:

1) The opendnp3 version

2) openssl version

3) Is this opendnp3 on both sides? Is the master also opendnp3?

4) If possible, a set of certificates for each side the recreates the problem

thanks,
Adam

vanjabucic, 2016-09-01:
1) The opendnp3 version
*2.1 official release*

2) openssl version
*openssl-1.0.1i-27.6.1  (SUSE SLES12 server)*

3) Is this opendnp3 on both sides? Is the master also opendnp3?
*Only our side is opendnp3. Not sure what the master side is or how TLS server is done*

4) If possible, a set of certificates for each side the recreates the problem
*OATI managed certs, we can't share*

Thanks,
Vanja

jadamcrain, 2016-09-01:
Thanks. As discussed on the list:

1) try to replicate using opendnp3 master. hopefully this will give us a clue if it is our problem.

2) Try to capture a wireshark trace and see if anything coming from the master looks out of sorts.

-AC

vanjabucic, 2016-09-01:
As a side-note, the master is PJMISO (Jetstream).

jadamcrain, 2016-10-20:
Just to record the history here, I believe that the new TLS functionality on the "development branch" should fix this issue.

closed 2016-11-10 by jadamcrain