dnp3/opendnp3 #175: Max Keep Alive Causes High CPU

spacecoder007, 2016-09-06:
Setting the keep alive timeout to TimeSpan.MaxValue on the master link causes high CPU.

This can be reproduced using the CLRMasterDemo and CLROutstationDemo for 2.2.0-M1.

spacecoder007, 2016-09-06:
OnKeepAliveTimeout is called immediately, debugging it shows the following:

Breakpoint at ASIOExecutor::Start:

time.milliseconds = 922338335267537
expiration = {_MyDur = {_MyRep=1131582059419200 }}

jadamcrain, 2016-09-06:
Could be some kind of overflow issue with the conversion from CLR TimeSpan to C++ type. I'll be able to take a peek when I return from holiday 09/18.

spacecoder007, 2016-09-06:
thanks Adam

jadamcrain, 2016-10-02:
This was happening in the C++ with TimeDuration::Max() as well. The underling issue was an overflow when converting from the opendnp3 representation (long of milliseconds) to the underlying representation (long of nanoseconds) resulting a negative timepoint and rapid fire of the timer.

This has been resolved in the development branch which will get merged into 2.0.x for the next 2.2.0 milestone release.

closed 2016-10-02 by jadamcrain

sidhoda, 2016-10-03:
Thanks Adam. I'll give it a test at next opportunity.

Looking through the commit, noticed a std::cout left in:
https://github.com/automatak/dnp3/blob/development/dotnet/bindings/CLRAdapter/src/Conversions.cpp#L427

jadamcrain, 2016-10-03:
Thanks for the review Sid. Removed.