dnp3/opendnp3 #177: Retrieve Channel/Link Statistics for IMasterSession

sidhoda, 2016-10-03:
Currently only stack statistics are accessible to listener accepted connections (IMasterSession), via the GetStackStatistics method. 

Ideally the following channel/link statistics would also be available for these connections:
- numCrcError
- numLinkFrameTx
- numLinkFrameRx
- numBadLinkFrameRx
- numBytesRx
- numBytesTx

sidhoda, 2016-10-22:
I was going to have a stab at adding this in. 

I was thinking the LinkChannelStatistics held on the LinkSession could be exposed via an accessor on the interface. IMasterSession/MasterSessionStack would then in turn have an accessor to retrieve the stats from the link session, so it can be brought back into the application.

Does that sound about right? Only thing I need to get my head around is how to do it in a thread safe way.

jadamcrain, 2016-10-22:
Use the same mechanism that's used to retrieve the other statistics:

https://github.com/automatak/dnp3/blob/development/cpp/libs/src/asiodnp3/MasterSessionStack.cpp#L123

Executor::ReturnFrom<T>(...) guaranteeds that the function is run on the strand and returned to the caller's thread.

Just be careful, as we currently drop the reference to the LinkSession when the stack goes offline:

https://github.com/automatak/dnp3/blob/development/cpp/libs/src/asiodnp3/MasterSessionStack.cpp#L74

Perhaps we can just make a copy of the statistics or something when it goes offline...

jadamcrain, 2016-10-25:
Be aware that I've refactored the way the statistics are handled in C++ on the development branch. Components now 'own' the statistics they increment, and there's no longer any passing pointers to them around. This should make things easier going forward.

closed 2020-04-17 by jadamcrain