dnp3/opendnp3 #192: Cross compile error

luzik, 2017-02-03:
Can't cross compile

> [  6%] Building CXX object CMakeFiles/asiopal.dir/cpp/libs/src/asiopal/Executor.cpp.o
> /opt/poky/2.0.2/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++   -march=armv5e -marm  -mthumb-interwork --sysroot=/opt/poky/2.0.2/sysroots/armv5e-poky-linux-gnueabi  -DASIO_STANDALONE -DOPENDNP3_USE_TLS -I/home/luzik/soft/dnp3/deps/asio/asio/include -I/home/luzik/soft/dnp3/./cpp/libs/src -I/home/luzik/soft/dnp3/./cpp/libs/include -I/home/luzik/soft/dnp3/./cpp/tests/libs/src  -Wall -Wno-multichar -Wunused-variable -Wno-unused-parameter -Wunused-function -Wunused -Wno-system-headers -Wwrite-strings -save-temps -fPIC -Wno-deprecated -Wno-parentheses -std=c++14 -O4 -DNDEBUG   -o CMakeFiles/asiopal.dir/cpp/libs/src/asiopal/Executor.cpp.o -c /home/luzik/soft/dnp3/cpp/libs/src/asiopal/Executor.cpp
> /home/luzik/soft/dnp3/cpp/libs/src/asiopal/Executor.cpp: In member function 'void asiopal::Executor::BlockUntil(const std::function<void()>&)':
> /home/luzik/soft/dnp3/cpp/libs/src/asiopal/Executor.cpp:97:21: error: aggregate 'std::promise<bool> ready' has incomplete type and cannot be defined
>   std::promise<bool> ready;
>

emgre, 2017-02-03:
It looks like your compiler does not support the full specification of C++11, specifically std::promise. I don't know arm-poky-linux-gnueabi, but do you have a RELEASE.txt for this specific compiler version ( something like https://launchpadlibrarian.net/287101683/release.txt ).

Some people had issues with another compiler GCC based ARM compiler not supporting std::promise [here](http://stackoverflow.com/questions/22036396/stdpromise-error-on-cross-compiling). The solution was to introduce the right compiler flags about the CPU. However, you already seem to include the flags about your CPU.

luzik, 2017-02-08:
I've updated gcc to edge 6.2 but still error in this place

Any hints ? What can I change ? 

/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++   -march=armv5e -marm --sysroot=/opt/poky/2.2.1/sysroots/armv5e-poky-linux-gnueabi  -DASIO_STANDALONE -DOPENDNP3_USE_TLS -I/home/luzik/soft/dnp3/deps/asio/asio/include -I/home/luzik/soft/dnp3/./cpp/libs/src -I/home/luzik/soft/dnp3/./cpp/libs/include -I/home/luzik/soft/dnp3/./cpp/tests/libs/src  -Wall -Wno-multichar -Wunused-variable -Wno-unused-parameter -Wunused-function -Wunused -Wno-system-headers -Wwrite-strings -save-temps -fPIC -Wno-deprecated -Wno-parentheses -std=c++14 -O4 -DNDEBUG   -o CMakeFiles/asiopal.dir/cpp/libs/src/asiopal/Executor.cpp.o -c /home/luzik/soft/dnp3/cpp/libs/src/asiopal/Executor.cpp
/home/luzik/soft/dnp3/cpp/libs/src/asiopal/Executor.cpp: In member function ‘void asiopal::Executor::BlockUntil(const std::function<void()>&)’:
/home/luzik/soft/dnp3/cpp/libs/src/asiopal/Executor.cpp:97:21: error: aggregate ‘std::promise<bool> ready’ has incomplete type and cannot be defined
  std::promise<bool> ready;
                     ^~~~~
CMakeFiles/asiopal.dir/build.make:113: polecenia dla obiektu 'CMakeFiles/asiopal.dir/cpp/libs/src/asiopal/Executor.cpp.o' nie powiodły się
make[2]: *** [CMakeFiles/asiopal.dir/cpp/libs/src/asiopal/Executor.cpp.o] Błąd 1
make[2]: Opuszczenie katalogu '/home/luzik/soft/dnp3/out-arm'
CMakeFiles/Makefile2:379: polecenia dla obiektu 'CMakeFiles/asiopal.dir/all' nie powiodły się
make[1]: *** [CMakeFiles/asiopal.dir/all] Błąd 2
make[1]: Opuszczenie katalogu '/home/luzik/soft/dnp3/out-arm'
Makefile:141: polecenia dla obiektu 'all' nie powiodły się
make: *** [all] Błąd 2
luzik@builder:out-arm$ /opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++ -v
Using built-in specs.
COLLECT_GCC=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++
COLLECT_LTO_WRAPPER=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/6.2.0/lto-wrapper
Target: arm-poky-linux-gnueabi
Configured with: ../../../../../../work-shared/gcc-6.2.0-r0/gcc-6.2.0/configure --build=x86_64-linux --host=x86_64-pokysdk-linux --target=arm-poky-linux-gnueabi --prefix=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr --exec_prefix=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr --bindir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi --sbindir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi --libexecdir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi --datadir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/share --sysconfdir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/etc --sharedstatedir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/com --localstatedir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/var --libdir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/lib/arm-poky-linux-gnueabi --includedir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/include --oldincludedir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/include --infodir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/share/info --mandir=/opt/poky/2.2.1/sysroots/x86_64-pokysdk-linux/usr/share/man --disable-silent-rules --disable-dependency-tracking --with-libtool-sysroot=/opt/oe-core4/build/tmp/sysroots/x86_64-nativesdk-pokysdk-linux --with-gnu-ld --enable-shared --enable-languages=c,c++ --enable-threads=posix --enable-multilib --enable-c99 --enable-long-long --enable-symvers=gnu --enable-libstdcxx-pch --program-prefix=arm-poky-linux-gnueabi- --without-local-prefix --enable-lto --enable-libssp --enable-libitm --disable-bootstrap --disable-libmudflap --with-system-zlib --with-linker-hash-style=gnu --enable-linker-build-id --with-ppl=no --with-cloog=no --enable-checking=release --enable-cheaders=c_global --without-isl --with-gxx-include-dir=/not/exist/usr/include/c++/6.2.0 --with-build-time-tools=/opt/oe-core4/build/tmp/sysroots/x86_64-linux/usr/arm-poky-linux-gnueabi/bin --with-sysroot=/not/exist --with-build-sysroot=/opt/oe-core4/build/tmp/sysroots/stgp3 --without-long-double-128 --enable-poison-system-directories --with-mpfr=/opt/oe-core4/build/tmp/sysroots/x86_64-nativesdk-pokysdk-linux --with-mpc=/opt/oe-core4/build/tmp/sysroots/x86_64-nativesdk-pokysdk-linux --disable-static --enable-nls --enable-initfini-array
Thread model: posix
gcc version 6.2.0 (GCC)

luzik, 2017-02-08:
It seems to be related with https://gcc.gnu.org/bugzilla/show_bug.cgi?id=64735 so std future will not compile on ARMv4 ARMv5 ..only ARMv7 and up. Fix was relased about month ago, so it will take some time to get it compile. Is it posibble to conditionally fix this issue somehow in your source code ?

jadamcrain, 2017-02-08:
Here's what the method does:

1) Submit an action (std::function<void ()>) to the executor to be run
2) Block the calling thread until the action has been processed on a thread-pool thread

You could use std::mutex / std::condition_variable to accomplish the same thing. Namely, you just need to way to block the calling thread and have it woken up once the action has been completed. std::promise was just the most efficient way to accomplish this in c++11.

luzik, 2017-02-09:
Is this safe workaround ?
```
diff --git a/cpp/libs/include/asiopal/Executor.h b/cpp/libs/include/asiopal/Executor.h
index ec725bc..d37b854 100644
--- a/cpp/libs/include/asiopal/Executor.h
+++ b/cpp/libs/include/asiopal/Executor.h
@@ -28,6 +28,7 @@
 #include "asiopal/SteadyClock.h"
 
 #include <future>
+#include "Synchronized.h"
 
 namespace asiopal
 {
@@ -97,6 +98,7 @@ T Executor::ReturnFrom(const std::function<T()>& action)
                return action();
        }
 
+       /*
        std::promise<T> ready;
 
        auto future = ready.get_future();
@@ -111,6 +113,19 @@ T Executor::ReturnFrom(const std::function<T()>& action)
        future.wait();
 
        return future.get();
+       */
+       else
+       {
+               Synchronized<T> sync;
+               auto pointer = &sync;
+               auto lambda = [action, pointer]()
+               {
+                       T tmp = action();
+                       pointer->SetValue(tmp);
+               };
+               strand.post(lambda);
+               return sync.WaitForValue();
+       }
 }
 
 
diff --git a/cpp/libs/src/asiopal/Executor.cpp b/cpp/libs/src/asiopal/Executor.cpp
index 6c3b70f..461789b 100644
--- a/cpp/libs/src/asiopal/Executor.cpp
+++ b/cpp/libs/src/asiopal/Executor.cpp
@@ -94,6 +94,7 @@ void Executor::BlockUntil(const std::function<void()>& action)
                return;
        }
 
+       /*
        std::promise<bool> ready;
 
        auto future = ready.get_future();
@@ -107,6 +108,19 @@ void Executor::BlockUntil(const std::function<void()>& action)
        strand.post(run);
 
        future.wait();
+       */
+       else
+       {
+               Synchronized<bool> sync;
+               auto pointer = &sync;
+               auto lambda = [action, pointer]()
+               {
+                       action();
+                       pointer->SetValue(true);
+               };
+               strand.post(lambda);
+               sync.WaitForValue();
+       }
 }
 
 void Executor::BlockUntilAndFlush(const std::function<void()>& action)
```

jadamcrain, 2017-02-09:
Yes. I believe so.

We had this custom solution for a time when there was a bug in early MSVC versions of std::promise/future.

-Adam

luzik, 2017-02-13:
Seems working fine. Thanks

closed 2017-02-13 by luzik

emgre, 2017-12-16:
I've just accidentaly hit this particular issue while trying mingw64. I'm using GCC 7.2.0 from mingw64, so I'm sure it supports future.

The fix was astonishingly simple: `#include <future>` at the beginning of Executor.cpp. The error mislead me thinking it was a bad compiler support, but perhaps it was not.