dnp3/opendnp3 #193: Compilation on Ubuntu Linux

hanusek, 2017-02-09:
Hello,
trying to compile openDNP3.

> git clone --recursive https://github.com/automatak/dnp3.git
> cmake ./dnp3 -DDNP3_DEMO=ON -DDNP3_TLS=ON

Unfortunately, I get an error. I put below the contents of the log file.
I have installed package **libpthread-stubs0-dev**.
Please help.

`Determining if the pthread_create exist failed with the following output:
Change Dir: /home/hans/automatak/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec4179527279/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec4179527279.dir/build.make CMakeFiles/cmTryCompileExec4179527279.dir/build
make[1]: Wejście do katalogu `/home/hans/automatak/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/hans/automatak/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec4179527279.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec4179527279.dir/CheckSymbolExists.c.o   -c /home/hans/automatak/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTryCompileExec4179527279
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec4179527279.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec4179527279.dir/CheckSymbolExists.c.o  -o cmTryCompileExec4179527279 -rdynamic 
CMakeFiles/cmTryCompileExec4179527279.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[1]: *** [cmTryCompileExec4179527279] Błąd 1
make[1]: Opuszczenie katalogu `/home/hans/automatak/CMakeFiles/CMakeTmp'
make: *** [cmTryCompileExec4179527279/fast] Błąd 2

File /home/hans/automatak/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/hans/automatak/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec4192076920/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec4192076920.dir/build.make CMakeFiles/cmTryCompileExec4192076920.dir/build
make[1]: Wejście do katalogu `/home/hans/automatak/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/hans/automatak/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec4192076920.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTryCompileExec4192076920.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-2.8/Modules/CheckFunctionExists.c
Linking C executable cmTryCompileExec4192076920
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec4192076920.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTryCompileExec4192076920.dir/CheckFunctionExists.c.o  -o cmTryCompileExec4192076920 -rdynamic -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [cmTryCompileExec4192076920] Error 1
make[1]: Opuszczenie katalogu `/home/hans/automatak/CMakeFiles/CMakeTmp'
make: *** [cmTryCompileExec4192076920/fast] Error 2`

jadamcrain, 2017-02-09:
"libpthread-stubs0-dev"

http://packages.ubuntu.com/trusty/libpthread-stubs0-dev

^ This package is definitely not required to build opendnp3.

What error did you get prior to installing this package?

The linker is having trouble finding "-lpthreads".

hanusek, 2017-02-09:
Log indicates that a problem with the pthreads.
Pthreads I have installed on system. Qt uses it and works OK.
Any ideas?

File **CMakeOutput.log** :
```
The system is: Linux - 3.13.0-105-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags: 

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/hans/automatak/CMakeFiles/2.8.12.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags: 

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/hans/automatak/CMakeFiles/2.8.12.2/CompilerIdCXX/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/hans/automatak/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec94124474/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec94124474.dir/build.make CMakeFiles/cmTryCompileExec94124474.dir/build
make[1]: Wejście do katalogu `/home/hans/automatak/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/hans/automatak/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec94124474.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec94124474.dir/testCCompiler.c.o   -c /home/hans/automatak/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTryCompileExec94124474
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec94124474.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec94124474.dir/testCCompiler.c.o  -o cmTryCompileExec94124474 -rdynamic 
make[1]: Opuszczenie katalogu `/home/hans/automatak/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/hans/automatak/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec2364804566/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec2364804566.dir/build.make CMakeFiles/cmTryCompileExec2364804566.dir/build
make[1]: Wejście do katalogu `/home/hans/automatak/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/hans/automatak/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec2364804566.dir/CMakeCCompilerABI.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec2364804566.dir/CMakeCCompilerABI.c.o   -c /usr/share/cmake-2.8/Modules/CMakeCCompilerABI.c
Linking C executable cmTryCompileExec2364804566
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec2364804566.dir/link.txt --verbose=1
/usr/bin/cc     -v CMakeFiles/cmTryCompileExec2364804566.dir/CMakeCCompilerABI.c.o  -o cmTryCompileExec2364804566 -rdynamic  
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.9/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.9.4-2ubuntu1~14.04.1' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.9.4 (Ubuntu 4.9.4-2ubuntu1~14.04.1) 
COMPILER_PATH=/usr/lib/gcc/x86_64-linux-gnu/4.9/:/usr/lib/gcc/x86_64-linux-gnu/4.9/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/4.9/:/usr/lib/gcc/x86_64-linux-gnu/
LIBRARY_PATH=/usr/lib/gcc/x86_64-linux-gnu/4.9/:/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../../lib/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-linux-gnu/4.9/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-o' 'cmTryCompileExec2364804566' '-rdynamic' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-linux-gnu/4.9/collect2 -plugin /usr/lib/gcc/x86_64-linux-gnu/4.9/liblto_plugin.so -plugin-opt=/usr/lib/gcc/x86_64-linux-gnu/4.9/lto-wrapper -plugin-opt=-fresolution=/tmp/ccLOkqh0.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --sysroot=/ --build-id --eh-frame-hdr -m elf_x86_64 --hash-style=gnu --as-needed -export-dynamic -dynamic-linker /lib64/ld-linux-x86-64.so.2 -z relro -o cmTryCompileExec2364804566 /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-g [... cut ...]
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTryCompileExec3551164192.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-2.8/Modules/CheckFunctionExists.c
Linking C executable cmTryCompileExec3551164192
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec3551164192.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTryCompileExec3551164192.dir/CheckFunctionExists.c.o  -o cmTryCompileExec3551164192 -rdynamic -lpthread 
make[1]: Opuszczenie katalogu `/home/hans/automatak/CMakeFiles/CMakeTmp'
```

jadamcrain, 2017-02-10:
You could try commenting out this find_package directive:

https://github.com/automatak/dnp3/blob/2.0.x/cmake/inc/posix.cmake#L3

hanusek, 2017-02-11:
OK. I solved the problem. I made a reinstallation g++-4.9 and gcc-4.9.
[AskUbuntu Gcc-4.9](http://askubuntu.com/questions/428198/getting-installing-gcc-g-4-9-on-ubuntu)

closed 2017-02-11 by hanusek