dnp3/opendnp3 #197: Outstation can get stuck in StateUnsolicitedConfirmWait forever

chk-mk, 2017-04-07:
I found when testing with the Triangle Microworks test harness that the outstation would appear to lock up if unsolicited responses were allowed.

1. On (re)connection, the outstation immediately sends a null unsol message
2. The master sends a disable unsolicited request at the same time
3. On receiving the null unsol, the test harness ignores the request (I'm not sure if this is permitted, but other masters may be doing it anyway, and it could happen by mistake over an unreliable connection):

> 09:37:13.849: **** mDNP   Discard Unsolicited Response in Startup state, waiting for Integrity Poll to complete **** 

4. On receiving the disable-unsolicited, the outstation cancels the unsol-confirm timer, responds to the master's request, but remains in StateUnsolicitedConfirmWait
5. It stays in this state forever since there's no timer and a confirm isn't coming

As far as I can see, the fundamental issue is that a non-read request received while waiting for an unsol reply effectively sets the timeout to infinite. I haven't seen this happen in other states, but any case where a timer is cancelled without changing state could have a similar issue.

My workaround for the moment is simply to leave the confirm timer running:

```
OutstationState& StateUnsolicitedConfirmWait::OnNewNonReadRequest(OContext& ctx, const APDUHeader& header, const openpal::RSlice& objects)
{
	ctx.deferred.Reset();
	//ctx.confirmTimer.Cancel();
	ctx.RespondToNonReadRequest(header, objects);
	return *this;
}
```

It will then time out and recover. I haven't noticed any side effects (yet). I'm happy with that, but I took a quick look at the standard (my old copy of it, at least) and it says not to expect a confirmation to an unsol request if a disable-unsol is received, so this may need to be a special case where it does cancel the timer but returns to idle.

jadamcrain, 2017-04-07:
Good find. Yes, the timer should not be canceled there. I'll write a unit test for this before fixing so we can prevent a future regression. Will issue 2.2.0-rc4 for this.

jadamcrain, 2017-04-08:
Wrote a test case and then removed the timer cancel:

9f339ec312f8b3e7916809cc6d46c63645b26897

closed 2017-05-01 by jadamcrain