dnp3/opendnp3 #198: Enable/Disable unsolcited not reseting after TCP reconnect

jadamcrain, 2017-04-17:
Reported by yesid8@gmail.com

neilstephens, 2018-01-30:
Gday,

I can't see anything in the DNP3 spec that implies that you would have to send Enable/Disable because of an interruption in comms.

The only thing I found that governs when to send those functions is when the outstation has restarted (or set the restart IIN flag to be precise. This makes sense to me since TCP (and everything below the link layer) is out of scope of the protocol anyway, so it shouldn't have much of a direct effect.

Just thought I'd chime in. I'd be interested to know if I've mis-read the spec.

Cheers,
Neil.

jadamcrain, 2018-05-10:
I think you're right.  These settings are supposed to persist across settings, and master are supposed to enable/disable after reconnecting anyway to ensure proper settings.

closed 2018-05-10 by jadamcrain