dnp3/opendnp3 #204: Read access error when using discontiguous indices

jadamcrain, 2017-07-07:
I'm having a bit of trouble with discontiguous points in open DNP3 slaves on Linux.  I've set the outstation up to use discontiguous points:

config.outstation.params.indexMode = IndexMode::Discontiguous;

I've then configured two binary inputs as follows:

for (int i = 0; i < 2; ++i)
{
    config.binary[i].clazz = PointClass::Class1;
    config.binary[i].svariation = StaticBinaryVariation::Group1Var2;
    config.binary[i].evariation = EventBinaryVariation::Group2Var3;
    config.binary[i].vIndex = 2 + i;
}

To send a change to the master, I do something like:

builder.Update(Binary(value, flags, time), 2 + i);

The values come through to the master on the right points, but at some point, valgrind complains of invalid memory access:

==1384== Thread 2:
==1384== Invalid read of size 2
==1384==    at 0x4EF3AFD: unsigned short opendnp3::Database::GetRawIndex<opendnp3::BinarySpec>(unsigned short) [clone .part.12] [clone .constprop.26] (in /usr/local/lib/libopendnp3.so)
==1384==    by 0x4EF3CE7: opendnp3::Database::Update(opendnp3::Binary const&, unsigned short, opendnp3::EventMode) (in /usr/local/lib/libopendnp3.so)
==1384==    by 0x5641F0B: asio::detail::completion_handler<asiodnp3::OutstationStack::Apply(asiodnp3::Updates const&)::{lambda()#1}>::do_complete(asio::detail::task_io_service*, asio::detail::task_io_service_operation*, std::error_code const&, unsigned long) (in /usr/local/lib/libasiodnp3.so)
==1384==    by 0x51B4954: asio::detail::strand_service::do_complete(asio::detail::task_io_service*, asio::detail::task_io_service_operation*, std::error_code const&, unsigned long) (in /usr/local/lib/libasiopal.so)
==1384==    by 0x519F5F1: asiopal::ThreadPool::Run(int) (in /usr/local/lib/libasiopal.so)
==1384==    by 0x5B41C7F: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21)
==1384==    by 0x58736B9: start_thread (pthread_create.c:333)
==1384==    by 0x61283DC: clone (clone.S:109)
==1384==  Address 0x739aff8 is 3,056,968 bytes inside an unallocated block of size 3,097,904 in arena "client"
==1384==

Am I using virtual point indices wrong?  I've tried using the plain indices (rather than virtual) for the updates, but then the points don't come through to the master on the correct indices.

Nick

nickelarse, 2017-07-07:
To reproduce, first set up two binary inputs as follows:
```
	for (int i = 0; i < 2; ++i)
	{
		config.binary[i].clazz = PointClass::Class1;
		config.binary[i].svariation = StaticBinaryVariation::Group1Var2;		// NOTE was Var1 on TMW; though Var2 is more appropriate as it has flags
		config.binary[i].evariation = EventBinaryVariation::Group2Var3;
		config.binary[i].vIndex = 2 + i;
	}
```
Then try and update an invalid binary input as follows:
```
		Flags flags;
		if (restart)
			flags.Set(BinaryQuality::RESTART);
		flags.Set(BinaryQuality::ONLINE);
			builder.Update(Binary(true, flags), 0, EventMode::Suppress);
```
When run under valgrind, you should get a message similar to the one posted in the bug report.

jadamcrain, 2017-07-07:
@nickelarse Thank you. I'll add a unit test that will make this easily reproducible on a single thread with the debugger running.

jadamcrain, 2017-09-22:
Resolved in 827cb6d4e26f14b7bd33f9d71a7f6d507fc5f1c8.

Thanks for the clear bug report.

closed 2017-09-22 by jadamcrain