dnp3/opendnp3 #210: Could NOT find JNI (missing: JAVA_AWT_LIBRARY JAVA_JVM_LIBRARY for RPI3

Malwaremamu, 2017-10-03:
pi@raspberrypi:~/dnp3 $ cmake ../dnp3 -DDNP3_TLS=ON
CMake build is: Release
GNU version:4.8.5
enabling all optional components
CMake Error at /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find JNI (missing: JAVA_AWT_LIBRARY JAVA_JVM_LIBRARY
  JAVA_INCLUDE_PATH JAVA_INCLUDE_PATH2 JAVA_AWT_INCLUDE_PATH)
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.7/Modules/FindJNI.cmake:305 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:46 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/dnp3/CMakeFiles/CMakeOutput.log".
See also "/home/pi/dnp3/CMakeFiles/CMakeError.log".
I have commented the lines in posix.cmake and while building I am getting error to find JNI.

Malwaremamu, 2017-10-03:
autoreconf --install
autoreconf: 'configure.ac' or 'configure.in' is required.

Please help. I am trying to resolve these from a month.

jadamcrain, 2017-10-03:
I think you need to delete your build directory and start over. It's caching another build where the Java 
bindings were enabled via -DDNP3_ALL=ON or -DDNP3_JAVA=ON.

The project doesn't use autotools. It uses cmake as described here:

https://www.automatak.com/opendnp3/docs/guide/current/

Please send any further problems with the build to the mailing list:

http://groups.google.com/group/automatak-dnp3

We only use github ticketing for bugs/features.

thanks,
Adam

closed 2017-10-03 by jadamcrain