dnp3/opendnp3 #213: C# Outstation leaks memory when loading measurement updates

jadamcrain, 2017-10-21:
https://github.com/automatak/dnp3/blob/2.0.x/dotnet/bindings/CLRAdapter/src/ChangeSetAdapter.h#L23

ChangeSetAdapter has no finalizer that frees native resource

jadamcrain, 2017-10-21:
Proposed fix in 915cc963efead09c2ae6273645068024474ae412

jadamcrain, 2018-01-12:
Reopening since there are reports that this still leaks. Need to do profiling to see why finalizer isn't being called.

samhightech, 2018-11-29:
I have been using opendnp3 using nuget in my C# Windows service, I update 5000 values each 5 seconds, and the ram is getting full in a crazy way, due to the changesets, please I need some guidance to fix this issue asap, the leakage is so obvious.

jadamcrain, 2018-11-29:
Try the 2.3.0-M1 release:

https://www.nuget.org/packages/opendnp3/2.3.0-M1

samhightech, 2018-12-06:
Yeah it minimized the memory leak, I have been investigating it for a while, thanks.

Also I want to ask about one off topic issue, regarding GI from my outstation, and time sync when asked by Master, how can I customize it, or is it already implemented?

emgre, 2019-05-15:
I was not able to find any leaks in the C# bindings in the 2.3.0-M1 release. I did some formating and a small optimisation in PR #276.

closed 2019-05-15 by emgre