dnp3/opendnp3 #214: C# Master leaks memory when calling Shutdown() and re-creating.

InFaMoUsZero, 2017-11-15:
The issue is due to the .NET Wrapper (DNP3ManagerAdapter) does not implement a finalizer.

Proposed fix:

DNP3ManagerAdapter.cpp
```javascript
DNP3ManagerAdapter::!DNP3ManagerAdapter()
{
   delete manager;
}

DNP3ManagerAdapter::~DNP3ManagerAdapter()
{
   this->!DNP3ManagerAdapter();
}
```

DNP3ManagerAdapter.h
```javascript
DNP3ManagerAdapter(System::Int32 concurrency, ILogHandler^ logHandler);
~DNP3ManagerAdapter();
!DNP3ManagerAdapter();

...
```

I haven't checked but if there needs to be cleanup done, then this is the pattern that needs to be applied or the Garbage Collector tends to ignore it completely, causing a leak.

jadamcrain, 2017-11-15:
Good catch @InFaMoUsZero.  We had a similar issue in one of the outstation adapter components recently. Want to submit a PR? If not, no biggie, I'll comb the issue list before the next RC.

InFaMoUsZero, 2017-11-15:
#215

jadamcrain, 2017-11-15:
Resolved in f2eaaad736f8469b01916de01ee44fadf80fa1d5

closed 2017-11-15 by jadamcrain