dnp3/opendnp3 #216: Java master hangs on shutdown

antiquechrono, 2017-11-29:
MacOS 10.12.6
Java version "1.8.0_91"
Java(TM) SE Runtime Environment (build 1.8.0_91-b14)
Java HotSpot(TM) 64-Bit Server VM (build 25.91-b14, mixed mode)

I pulled and built the 2.0.x branch last night and was messing with the Java master-demo. When you type in quit it prints the following and never quits.

Channel state: CLOSED
Channel state: SHUTDOWN

I have not changed any of the code in the master-demo and am running it via 
mvn -pl example exec:java@master-demo

In my own code I have even tried calling System.exit(0) after master.shutdown() and it still hangs. If I remove the call to master.shutdown() then everything exits as expected.

I did a little profiling and it looks like it's getting stuck in com.automatak.dnp3.impl.ManagerImpl.shutdown_native_manager[native]() and never exiting. I've never used JNI before so I'm not sure where this code lives to debug any further.

Also the outstation-demo exits correctly.

I was going to test this with the 2.2.0 tagged branch but it errors out compiling asiodnp3/ErrorCodes.cpp. If I had to guess this is probably due to the fact that Apple's Clang compiler is awful. I can report this in another issue if you want.

jadamcrain, 2017-11-29:
Thanks for the report. I won't have time to test this out until week  12/11.  

That JNI method invokes the destructor for the C++ DNP3 Manger.  That calls terminates everything in an ordered way, and then calls join() for each thread in the thread pool.  It sounds like something is not cleaning up properly.

Do you get the same results if you run the C++ master demos on MaxOS?

antiquechrono, 2017-12-14:
Sorry about the delay, I got really busy because of moving. When I try the C++ master and outstation they both exit cleanly.

jadamcrain, 2017-12-16:
Version of Clang? I can try this version on Ubuntu and see if I can replicate.

antiquechrono, 2017-12-16:
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.1.0 (clang-802.0.42)
Target: x86_64-apple-darwin16.7.0
Thread model: posix

ffreckle, 2018-04-26:
I'd like to report a similar issue using a [Python wrapper library](https://github.com/Kisensum/pydnp3) that was created with pybind11.  We've got a working Python outstation and master.  Everything works fine except for shutting down the outstation.  I've traced the call to DNP3Manager.Shutdown() and see the channel calling its ShutdownImpl and the MasterStack resource being freed but something is keeping io_service::run() blocked.  One theory is that shared pointer references in Python are keeping certain destructors from getting called but it looks like the call to DNP3Manager.Shutdown() should invoke all the shutdown code that is also called from destructors in classes like DNP3Channel and MasterStack.  Any ideas about where I might look or how I might determine what is causing the run() method to block?

jadamcrain, 2018-04-26:
It is probably caused by the same underlying issue in the C++, that is somehow surfaced by both bindings. I plan to review this ticket when I have time in mid-May.

jadamcrain, 2018-05-10:
Update: I can replicate this on Linux.

ffreckle, 2018-06-08:
The hang on shutdown problem occurring in our [pydnp3](https://github.com/Kisensum/pydnp3) test code appears to be due to python references to opendnp3 objects, specifically the `channel` returned by `manager.AddTCPClient(...)` and the `master` returned by `channel.AddMaster(...)`, perhaps this is not surprising.  If we explicitly delete these references before calling `manager.Shutdown()`, everything works as expected.  I think the Python bindings increment a shared_ptr reference on these objects.  That may be interfering with a shutdown process that expects the destructors of these to objects to be executed.

jadamcrain, 2018-07-09:
A user reported a similar issue in the C++ under the right conditions, which I have been able to resolve in this branch:

https://github.com/automatak/dnp3/tree/bug/deadlock_and_lifetimes

An infinite timer was being created that never exited, and this blocked the thread pool thread from exiting cleanly.

I just tested the master demo, and it also resolves this issue as well. This branch will get merged in 2.0.x for the upcoming 2.3.0-RC1 release.

neilstephens, 2019-06-11:
Gday,

The merged fixes definitely improved the situation, but I can still produce a deadlock on cleanup using this example application built against 2.3.0 (700dc961d). And it crops up in the wild in my application tests for [opendatacon](https://github.com/neilstephens/opendatacon).

[main.cpp.txt](https://github.com/dnp3/opendnp3/files/3274196/main.cpp.txt)

The main thread gets stuck in ThreadPool::Shutdown(), blocked on joining the threads. The rest of the threads are still in ThreadPool::Run. So there must still be some asio task that's keeping everything from exiting.

I'll have a go at identifying the offending asio task by turning on asio debug output, but I haven't had much luck trying to interpret that output before. So maybe you guys can take another look?

Regards,
Neil.

jadamcrain, 2019-06-11:
Thanks Neil. @emgre dived the ASIO debug mode to fix some of the issues, so shouldn't be a problem finding the issue.

neilstephens, 2019-06-11:
ASIO debug mode put me straight onto the source of the problem:

After it locked up, I could see that there was still a post, callback cycle happening every few seconds, so I was able to just put a breakpoint in asiopal::Executor::Post(), and sure enough, there is a SocketChannel re-transmit cycle that is stopping asio run from exiting.

[image]

I can see that asiodnp3::IOHandler::isShutdown bool is still false, so I'm assuming that is the problem. ie. Shutdown() on the SocketChannel hasn't been called in the shutdown sequence. How to fix it... that's where I get lost. Over to you :-)

Regards,
Neil.

neilstephens, 2019-06-11:
I think I've given you a bum steer, sorry.

The example code is using two separate DNP3Managers (I've forgotten why I did that now - probably just ruling things in/out when I was troubleshooting the original problem).

So the re transmit cycle I was observing was on the other DNP3Manager. The DNP3Managers each have a separate asio::io_context, so they shouldn't impact each other shutting down... so I'm back to not knowing what's stopping asio run from returning.

Regards,
Neil.

emgre, 2019-06-11:
@neilstephens On what platform/compiler are you experiencing this? I compiled your `main.cpp` on my Windows machine using MSVC 2019, both Debug and Release, and the program looks fine. It doesn't hang and use a constant amount of memory.

emgre, 2019-06-12:
@neilstephens 
cc @jadamcrain 

I think I found the issue. With `ASIO_ENABLE_HANDLER_TRACKING` enabled, I was able to track down the handler that was preventing the thread pool to finish. It's in [TCPServer](https://github.com/dnp3/opendnp3/blob/master/cpp/libs/src/asiopal/TCPServer.cpp).

On Windows, if an acceptor already accepted a connection, putting the callback in ASIO handler queue, and you close the acceptor before being called back, the accepted socket becomes invalid. I added [this check](https://github.com/dnp3/opendnp3/blob/master/cpp/libs/src/asiopal/TCPServer.cpp#L112-L118) a [few commits ago](https://github.com/dnp3/opendnp3/commit/cb97034bbb1a494b0fac55b6105d942acbcbd908#diff-998cc72e85363c61abd997931be1f472) to prevent the fault.

On Linux, under the same circumstances, the socket is simply reported as valid. Therefore, the socket is accepted and the accept loop is continued even if the server was shutdown.

I'm working on a fix at the moment, I'll let you know when it's ready. It also affects the TLS server.

Thanks for the report!

emgre, 2019-06-12:
You can try branch [`bugfix-tcpserver-shutdown`](https://github.com/dnp3/opendnp3/tree/bugfix-tcpserver-shutdown) and tell us if it fixes your issue.

neilstephens, 2019-06-12:
Cool! I'll try out that branch in the next couple of hours. Sorry I didn't get around to forwarding those emails, but looks like you had enough info to go off anyway. Thanks.

neilstephens, 2019-06-12:
Seems fixed to me. Nice work.

I'll have to look up some tutorials on using ASIO_ENABLE_HANDLER_TRACKING. It seems really powerful, and really neat how you can visualise what's going on with the handlerviz.pl script. I just have to figure out how to interpret the graphs and link the info back to the code.

Cheers,
Neil.

emgre, 2019-06-17:
The fixes were merge in release [2.3.1](https://github.com/dnp3/opendnp3/releases/tag/2.3.1).

PR #265 (2019-05-09): Fix deadlock and lifetime issues
Should fix #216 and #228.

There were a couple of issues in the code:
- `MasterSchedulerBackend::Shutdown` was never called. Therefore, some timers were not cancelled and the application would hang. I added the necessary call in `DNP3Channel::ShutdownImpl` and `MasterSessionStack::BeginShutdown`.
  - By the way, `MasterSessionStack::BeginShutdown` was never called. It is now called by `LinkSession::ShutdownImpl`.
- In `TCPClientIOHandler::StartConnect`, a timer was started when a connection failed, even if the connection the IO handler was shutdown. This timer  [... cut]
PR #310 (2019-06-17): Fix TCPServer deadlock (master branch)
Closes #216 on the master branch.

I'm not sure how we'll handle bugfix in the master branch.
closed 2019-06-17 by emgre