dnp3/opendnp3 #217: DNP Time is not translated properly onto the packet.

sanhardik, 2017-12-06:
Hello, 
I am trying to set the DNP time as epoch time in milliseconds. The counter is defined as Group 22 variation 5
`
uint64_t temp_time = 1512595515000;
builder.Update(Counter(mr_state.counter, 0x01, DNPTime(temp_time)), 0);
`
Time being set: 1512595515000 ms (Hex: 0x‭1602DB7E678‬)

In the packet capture on Wireshark it shows up as Jan 9, 1970 21:03:46.808 UTC  (Hex: 78 e6 b7 2d 00 00) 
It seems to be dropping 0x01 0x60

jadamcrain, 2017-12-08:
I was unable to replicate with unit tests:

https://github.com/automatak/dnp3/commit/926ee12df3db47a8cb9f1834bb22ff067af9da15

Can you upload a PCAP along with information about your platform/compiler?

thanks,
Adam

sanhardik, 2017-12-17:
It was a mistake in the code that was causing this. 
Had to use uint64_t as Time data type, multiply it by 1000 to get milliseconds and then use DNPTime(variable) to get the time.

closed 2017-12-17 by sanhardik