dnp3/opendnp3 #221: Outstation-demo using consistently high CPU

chintanhshah, 2018-01-29:
Hi @jadamcrain 
I have been able to compile and run the DNP3 outstation on Ubuntu12.4 on our test environment . However , I am noticing outstation-demo consistently using >98% CPU . This is slowing down the entire machine and sometimes making it inaccessible . 

Here is the output of the "top" on Ubuntu : 

# 27054 root      20   0 98952 2332 2040 R 98.6  0.1   0:21.54 outstation-demo
 2574 root      20   0 1529m  92m 1704 S  0.3  4.6   7:55.58 java
10572 root      20   0 21856 5784 3500 S  0.3  0.3 122:21.09 notify
23981 root      20   0 64464 2696 1996 S  0.3  0.1   0:00.65 sshd
27073 root      20   0 17392 1308  956 R  0.3  0.1   0:00.01 top
    1 root      20   0 24968 2668 1208 S  0.0  0.1   0:01.28 init
    2 root      20   0     0    0    0 S  0.0  0.0   0:00.00 kthreadd

If I do ltrace on this machine , I see the following : 

_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40
_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40
_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40
_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40
_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40
_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40
_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40
_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40
_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40
_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40
_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40
_ZNK8asiodnp313UpdateBuilder5BuildEv(0x7fff09fb4b40, 0x7fff09fb4b50, 0x7fff09fb4b10, 0x7f55846b2000, 0x7f5583341970) = 0x7fff09fb4b40

To me , this is most likely the infinite loop code in the outstation/main.cpp . Is this somethething that has been observed before and can be fixed in the existing code . 

Regards

jadamcrain, 2018-01-31:
That's an odd result. This call should block:

https://github.com/automatak/dnp3/blob/2.0.x/cpp/examples/outstation/main.cpp#L117

Can you run this with a debugger an get more info?

jadamcrain, 2018-03-06:
Any debugger info or sample executable? Otherwise I will need to close this issue.

closed 2018-04-26 by jadamcrain

chintanhshah, 2018-04-26:
Hi Adam

It's Ok to close it for now..Will check this out later if it comes back..

Thanks
Chintan

jadamcrain, 2018-04-26:
Thanks. Didn't hear back so I assumed you hadn't been able to replicate.