dnp3/opendnp3 #227: openssl 1.1 incompatibility, invalid use of incomplete type ‘X509 {aka struct x509_st}’

vanjabucic, 2018-05-03:
Compilation against openssl 1.1+ is failing:

```
/dnp3/cpp/libs/src/asiodnp3/tls/MasterTLSServer.cpp:83:17: error: invalid use of incomplete type ‘X509 {aka struct x509_st}’
      RSlice(cert->sha1_hash, SHA_DIGEST_LENGTH), // the thumbprint
                 ^~
In file included from /usr/include/openssl/crypto.h:31:0,
                 from /usr/include/openssl/bio.h:20,
                 from /usr/include/openssl/conf.h:13,
                 from /dnp3/deps/asio/asio/include/asio/ssl/detail/openssl_types.hpp:19,
                 from /dnp3/deps/asio/asio/include/asio/ssl/context_base.hpp:19,
                 from /dnp3/deps/asio/asio/include/asio/ssl/context.hpp:27,
                 from /dnp3/deps/asio/asio/include/asio/ssl.hpp:19,
                 from /dnp3/./cpp/libs/src/asiopal/tls/SSLContext.h:29,
                 from /dnp3/./cpp/libs/src/asiopal/tls/TLSServer.h:32,
                 from /dnp3/./cpp/libs/src/asiodnp3/tls/MasterTLSServer.h:26,
                 from /dnp3/cpp/libs/src/asiodnp3/tls/MasterTLSServer.cpp:22:
/usr/include/openssl/ossl_typ.h:119:16: note: forward declaration of ‘X509 {aka struct x509_st}’
 typedef struct x509_st X509;
                ^~~~~~~
make[2]: *** [CMakeFiles/asiodnp3.dir/build.make:567: CMakeFiles/asiodnp3.dir/cpp/libs/src/asiodnp3/tls/MasterTLSServer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:181: CMakeFiles/asiodnp3.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

jadamcrain, 2018-05-05:
I believe this was resolved by this pull request:

https://github.com/automatak/dnp3/pull/207

This is not part of a stable release yet. Can you try and confirm?

vanjabucic, 2018-05-07:
Compiled successfully using 2.0.x branch.

closed 2018-05-07 by jadamcrain