dnp3/opendnp3 #231: Error loading libasiodnp3.so.2 when running Outstation example

alexhennieroed, 2018-05-16:
I am working on an experiment running this library on Raspberry Pi 3 Model Bs. I currently have two set up; one is running the Master example and one is running the Outstation example. I am trying to add a second Pi running the Outstation example, but keep running into an issue where I successfully compile the code on the device with
`g++ -Wall -o outstation2 main.cpp -lasiodnp3 -lasiopal -lopendnp3 -lopenpal -lpthread`,
but running `./outstation2` results in
`./outstation2: error while loading shared libraries: libasiodnp3.so.2: cannot open shared object file: No such file or directory`
However, the `libasiodnp3.so.2` file is exactly where it should be (where it is located on the other two devices), and I installed ASIO and opendnp3 on this Pi the same way I did on the other two (clone ASIO into a directory and recursive clone opendnp3 into the same directory, then follow the cmake and Linux install steps). g++ and gcc are both version `Raspbian 6.3.0-18+rpi1+deb9u1` on all three devices. Neither of the other two devices have experience any issues running the library.
Could this be an issue with this specific device, an issue with my install, or something else?

jadamcrain, 2018-05-18:
run `ldd` on the binary and share the results.

alexhennieroed, 2018-05-22:
Sorry about the delay; here's the results:
`~/Desktop/opendnp3/dnp3 $ ldd libasiodnp3.so.2`
	`linux-vdso.so.1 (0x7ee03000)`
	`/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76e45000)`
	`libasiopal.so.2 => ~/Desktop/opendnp3/dnp3/libasiopal.so.2 (0x76dec000)`
	`libopendnp3.so.2 => ~/Desktop/opendnp3/dnp3/libopendnp3.so.2 (0x76cee000)`
	`libopenpal.so.2 => ~/Desktop/opendnp3/dnp3/libopenpal.so.2 (0x76cd7000)`
	`libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x76b79000)`
	`libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76afa000)`
	`libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76acd000)`
	`libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x7698e000)`
	`/lib/ld-linux-armhf.so.3 (0x76ee9000)`

jadamcrain, 2018-05-27:
Sorry, I should have been more specific. Can you run ldd on the executable?

```ldd outstation2```

alexhennieroed, 2018-07-17:
Sorry for the extended delay; I've been away for a while. Here's the output:
`~/Desktop/opendnp3/pi_code $ ldd outstation2`
	`linux-vdso.so.1 (0x7efdb000)`
	`/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76ee6000)`
	`libasiodnp3.so.2 => /usr/local/lib/libasiodnp3.so.2 (0x76e42000)`
	`libasiopal.so.2 => /usr/local/lib/libasiopal.so.2 (0x76de9000)`
	`libopendnp3.so.2 => /usr/local/lib/libopendnp3.so.2 (0x76ceb000)`
	`libopenpal.so.2 => /usr/local/lib/libopenpal.so.2 (0x76cd4000)`
	`libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76cab000)`
	`libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x76b63000)`
	`libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76ae4000)`
	`libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76ab7000)`
	`libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76978000)`
	`/lib/ld-linux-armhf.so.3 (0x76efc000)`

I attempted to run the program today and it worked this time. Nobody had touched the Pi or changed the code, so I'm not sure what caused it to work but I'm glad it sorted itself out. I'm not sure if the `ldd` will have any answers, but I figured I'd include it just in case.

closed 2018-07-17 by alexhennieroed

dunn1013, 2019-09-23:
Hi Adam,

I'm having the same problem.

sn300A381700008:~/ger_outstation$ ldd ger_outstation
        linux-vdso.so.1 (0x7ee91000)
        libopendnp3.so.2 => not found
        libopenpal.so.2 => not found
        libasiodnp3.so.2 => not found
        libasiopal.so.2 => not found
        libpthread.so.0 => /lib/libpthread.so.0 (0x76f60000)
        librt.so.1 => /lib/librt.so.1 (0x76f49000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x76e08000)
        libm.so.6 => /lib/libm.so.6 (0x76d8e000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x76d62000)
        libc.so.6 => /lib/libc.so.6 (0x76c26000)
        /lib/ld-linux-armhf.so.3 (0x76f89000)

emgre, 2019-09-23:
You need to put the dynamic libraries (libopendnp3.so, libopenpal.so, etc.) in a recognized library path of your system (e.g. `/usr/local/lib`), or right next to the executable.