dnp3/opendnp3 #232: Disable unsolicited and Integrity after reset connection

mosseedwin, 2018-05-18:
If the connection is reset, the master does not execute the disable unsolicited and integrity. These options are enabled in the master and only run the first time you connect to the slave.
Also when it is connecting again to the slave sends the messages with restarted flag active.

MasterConfig
        public bool disableUnsolOnStartup;
        public ClassField startupIntegrityClassMask;

jadamcrain, 2018-05-18:
Will have a look before 2.3.0-RC1 which is pending.

emgre, 2019-05-13:
The `DISABLE_UNSOLICITED` is required only if the master is restarted, not when the communication is lost, not if the outstation has the `DEVICE_RESTART` IIN set. Therefore, if you set the `disableUnsolOnStartup` to `true`, the expected behaviour is that the master **won't** send a `DISABLE_UNSOLICITED` when the connection is reset.

The startup integrity is only necessary if the master is restarted or if the outstation sets the `DEVICE_RESTART` IIN flag. In the case of a communication lost, chances are the outstation stayed alive and did not set the `DEVICE_RESTART` flag. In that case, it is not necessary to issue an integrity scan.

See IEEE 1815-2012 section 5.1.1.1.2 (p. 101-102) for all the details about the master expected behaviour when an outstation is restarted.

Here are the validations I did with the master branch:
- Master does **not** send `DISABLE_UNSOLICITED` and does **not** perform an integrity scan on communication lost with an outstation.
- Master does **not** send `DISABLE_UNSOLICITED` and does perform an integrity scan if the outstation sets the `DEVICE_RESTART` IIN flag.
- Master does send `DISABLE_UNSOLICITED` and does perform an integrity scan when it is restarted.

closed 2019-05-13 by jadamcrain