dnp3/opendnp3 #236: outstation cpp example

ncasaleiro, 2018-06-07:
config.analog[0].svariation = StaticAnalogVariation::Group30Var5;
	config.analog[0].evariation = EventAnalogVariation::Group32Var7;

With Group30Var5   doesn't send timestamp. But with Group30Var1 will send timestamp. I think It is a bug. 
But I don't have much experience under DNP3.

ncasaleiro, 2018-06-07:
when occurring at an event. By polling sends timestamp.

jadamcrain, 2019-05-10:
The issue is that the correct way to assign the default variation is:

```
db.analogs[0].config.evariation = EventAnalogVariation::Group32Var5;
```

or

```
db.analogs[0].config.evariation = EventAnalogVariation::Group32Var7;
```

Not

```
db.analogs[0].event.evariation = EventAnalogVariation::Group32Var5;
```

This is very confusing and we plan to improve the API in the next major release.

closed 2019-05-10 by jadamcrain