dnp3/opendnp3 #238: TaskConfig::With taking a reference with an undetermined lifetime

emgre, 2018-06-12:
[This](https://github.com/automatak/dnp3/blob/e00ff31b837821064e5208c15866a9d46f8777b1/cpp/libs/include/opendnp3/master/TaskConfig.h#L51-L54) member function is taking a reference to a `ITaskCallback` and only stores a pointer of it. However, this callback must live for the whole request/response lifetime.

A primitive:
```
{
     FooTaskCallback cb{} // inherits from ITaskCallback
     master->ScanRange(opendnp3::GroupVariationID(30, range.variation), 0, 10, opendnp3::TaskConfig::With(cb));
}
```
will segfault due to the callback going out of scope.

The `ITaskCallback` should be taken by `shared_ptr` in order to correct the issue.

emgre, 2018-06-12:
It would also be nice to have a wrapper that executes a lambda on the `OnComplete`. I have implemented my own like that:

```
#include "opendnp3/master/ITaskCallback.h"

class LambdaTaskCallback : public opendnp3::ITaskCallback
{
public:
    LambdaTaskCallback(std::function<void(opendnp3::TaskCompletion)> cb)
        : m_callback{cb}
    {}

    void OnStart() override {}
    void OnDestroyed() override {}

    void OnComplete(opendnp3::TaskCompletion result) override
    {
        m_callback(result);
    }

private:
    std::function<void(opendnp3::TaskCompletion)> m_callback;
};
```

This is useful for one-time requests and provides similar usage as `ICommandProcessor` where we pass a simple `std::function<void(const ICommandTaskResult&)>`.

jadamcrain, 2019-05-24:
We'll take  a comprehensive look at the way all of the master API methods make use of callbacks in 3.0.0.

emgre, 2020-04-27:
The callback handler is taken by `shared_ptr` now.

commit 28de3859 (2020-04-20): TaskConfig takes a shared_ptr to the callbacks. Fix #238.
commit 8e0b763d (2020-04-23): TaskConfig takes a shared_ptr to the callbacks. Fix #238.
PR #367 (2020-04-27): Fix remaining items for 3.0 release
- Fix #290: the only lib that was still exposed publicly was `log4cpp`. Since it's small and not really used elsewhere, it was copied right into opendnp3 and all references to it was removed. The library doesn't have any public dependency 🎉 (except for OpenSSL when compiling with TLS support).
- Fix #238: `TaskConfig` now takes a `std::shared_ptr` instead of a reference.
- Fix #293: Instead of having an out-parameter for the result of `DNP3Manager` calls, it now throws a `DNP3Error`.
  - All methods of `DNP3Manager` can throw an exception, notably because th [... cut]
closed 2020-04-27 by emgre