dnp3/opendnp3 #240: Unable to scan range group 0 for any variation.

henkbredell, 2018-07-06:
Hi,
I am unable to scan the object/group 0 for any of the implemented variations on my device.  I tested with another DNP3 master emulator and the device does work correctly, just not with the opendnp3 code.
I get:
ms(1530881829093) --AL->  TCV300002 - C1 00 
ms(1530881829093) --AL->  TCV300002 - FIR: 1 FIN: 1 CON: 0 UNS: 0 SEQ: 1 FUNC: CONFIRM
ms(1530881829093) INFO    TCV300002 - Begining task: Enable Unsolicited
ms(1530881829093) --AL->  TCV300002 - C2 14 3C 02 06 3C 03 06 3C 04 06 
ms(1530881829093) --AL->  TCV300002 - FIR: 1 FIN: 1 CON: 0 UNS: 0 SEQ: 2 FUNC: ENABLE_UNSOLICITED
ms(1530881829093) --AL->  TCV300002 - 060,002 - Class Data - Class 1 - all objects
ms(1530881829093) --AL->  TCV300002 - 060,003 - Class Data - Class 2 - all objects
ms(1530881829093) --AL->  TCV300002 - 060,004 - Class Data - Class 3 - all objects
ms(1530881829691) WARN    tcpclient - Skipped 13 bytes seaching for start bytes
ms(1530881830601) <-AL--  TCV300002 - FIR: 1 FIN: 1 CON: 0 UNS: 0 SEQ: 2 FUNC: RESPONSE IIN: [0x00, 0x00]
ms(1530881830601) INFO    TCV300002 - Begining task: Application Poll
ms(1530881830601) --AL->  TCV300002 - C3 01 00 FF 01 00 00 00 00 
ms(1530881830601) --AL->  TCV300002 - FIR: 1 FIN: 1 CON: 0 UNS: 0 SEQ: 3 FUNC: READ
ms(1530881830601) WARN    TCV300002 - **Unknown object 0 / 255**
ms(1530881831361) <-AL--  TCV300002 - FIR: 1 FIN: 1 CON: 0 UNS: 0 SEQ: 3 FUNC: RESPONSE IIN: [0x00, 0x04]

henkbredell, 2018-07-06:
here is a code snippet from where I am trying to make the call.
	struct class_scans_info {
		int 		device_ind;				// device index
		std::string type;			// type of scan to perform
		std::string scope;			// scope of class scan operation
		uint8_t 	group;					// group for range scan
		uint8_t 	variation;				// variation for range scan
		uint16_t 	start;
		uint16_t	stop;
		std::string interval_unit;	// units for timer interval
		int 		interval_qty;			// time 
		std::shared_ptr<IMasterScan> channel;
	};
	
	...
	
	scans[i].start and scans[i].stop are assigned uint8_t values 0 and 0
	scan_int is the scan interval in seconds
	
	...

opendnp3::GroupVariationID gv;
gv.group = scans[i].group;
gv.variation = scans[i].variation;
scans[i].channel = mstr[dev_ind].master->AddRangeScan(gv, scans[i].start, scans[i].stop, TimeDuration::Seconds(scan_int));

jadamcrain, 2018-07-06:
You can find a description of the features supported by opendnp3 here:

https://www.automatak.com/opendnp3/docs/guide/current/features/features/#compliance

Device attributes (group 0) are not supported at this time.

closed 2018-07-06 by jadamcrain