dnp3/opendnp3 #256: When using the java api variation type

borissmidt, 2019-02-14:
When i scan using the java api the variation isn't picked up by the master.
I wanted to scan for group 32 (analog) variation 6 (double) but when i run the folowing code it only scans for variation 1. i tried version 2.2.0 and the current master version both seem to have the issue.

```
master.addPeriodicScan(Duration.ofMillis(1),util.Arrays
    .asList(
      Header.allObjects(32.toByte, 6.toByte)
    ))
```

jadamcrain, 2019-02-14:
Thanks, added to the list of things to investigate before next release

borissmidt, 2019-02-14:
I will check the older versions tomorrow.

emgre, 2019-05-13:
I was not able to reproduce the issue on the master branch, nor the 2.2.0 release. Adding a periodic scan for g32v6 with the Java bindings works. It reports the correct variation in the logs and Wireshark also confirms that the correct variation is used.

closed 2019-05-13 by jadamcrain