dnp3/opendnp3 #257: Build with SSL fails with OPENSSL_ROOT_DIR specified

msqr, 2019-02-14:
Hello, I am trying to build on macOS using homebrew-installed OpenSSL. My `cmake` command I'm trying is this, with some relevant output:

```sh
$ cmake ../.. -DDNP3_JAVA=ON -DDNP3_TLS=ON -DOPENSSL_ROOT_DIR=/usr/local/opt/openssl -DOPENSSL_LIBRARIES=/usr/local/opt/openssl/lib

-- The C compiler identification is AppleClang 10.0.0.10001044
-- The CXX compiler identification is AppleClang 10.0.0.10001044
...
ASIO has been checked out as a git submodule: dnp3/deps/asio/asio/include
-- Found OpenSSL: /usr/local/opt/openssl/lib/libcrypto.dylib (found version "1.0.2q")  
OpenSSL libraries: /usr/local/opt/openssl/lib/libssl.dylib;/usr/local/opt/openssl/lib/libcrypto.dylib
...
```

So OpenSSL is found by `cmake` and then compiling starts off OK, but then fails when `asio` tries to include `<openssl/config.h>`:

```
[ 11%] Building CXX object CMakeFiles/asiopal.dir/cpp/libs/src/asiopal/tls/SSLContext.cpp.o
In file included from /Users/matt/Documents/SolarNetwork/Developer/git/dnp3/cpp/libs/src/asiopal/tls/SSLContext.cpp:22:
In file included from /Users/matt/Documents/SolarNetwork/Developer/git/dnp3/./cpp/libs/src/asiopal/tls/SSLContext.h:29:
In file included from /Users/matt/Documents/SolarNetwork/Developer/git/dnp3/deps/asio/asio/include/asio/ssl.hpp:19:
In file included from /Users/matt/Documents/SolarNetwork/Developer/git/dnp3/deps/asio/asio/include/asio/ssl/context.hpp:27:
In file included from /Users/matt/Documents/SolarNetwork/Developer/git/dnp3/deps/asio/asio/include/asio/ssl/context_base.hpp:19:
/Users/matt/Documents/SolarNetwork/Developer/git/dnp3/deps/asio/asio/include/asio/ssl/detail/openssl_types.hpp:19:10: fatal error: 'openssl/conf.h' file not found
#include <openssl/conf.h>
         ^~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/asiopal.dir/cpp/libs/src/asiopal/tls/SSLContext.cpp.o] Error 1
make[1]: *** [CMakeFiles/asiopal.dir/all] Error 2
make: *** [all] Error 2
```

Running with `VERBOSE=1` I don't see any `-I/usr/local/opt/openssl/include` in the build path, as expected. Do you know how to get this path included, or some other change I can make, for the build to succeed?

jadamcrain, 2019-02-19:
I wish I had a clear answer for you, but my guess this more a question for the folks that develop/maintain the CMake scripts for finding OpenSSL on your platform. It is peculiar that the libraries are "found", but that the relevant variables are presumably empty?

You might try printing some of them using message(...) during CMake initialization and see if they are indeed empty. That will at least help us narrow down the issue.

msqr, 2019-02-20:
I poked around in `CMakeLists.txt` and found that the found OpenSSL include directory was only added to the search path on Windows platforms. If I changed it like this the compile works for me:

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 8050772a7..cedbb7539 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -77,9 +77,7 @@ if(DNP3_TLS)
 
        find_package(OpenSSL REQUIRED)
        message("OpenSSL libraries: ${OPENSSL_LIBRARIES}")
-       if(WIN32)
-               include_directories(${OPENSSL_INCLUDE_DIR})
-       endif()
+       include_directories(${OPENSSL_INCLUDE_DIR})
 endif()
 
 set(CMAKE_REQUIRED_FLAGS ${CMAKE_CXX_FLAGS})
```

jadamcrain, 2019-02-20:
Excellent. I'll keep this in mind as we have plans to modernize the CMake project before the next major release.

jadamcrain, 2019-05-14:
Fixed in https://github.com/automatak/dnp3/commit/d9aab30985a0a97dcfc76f7f5735ee85fcba214e

closed 2019-05-14 by jadamcrain