dnp3/opendnp3 #261: Reimplement discontiguous index mode

jadamcrain, 2019-03-07:
The API allows for non-valid configurations. It would be way better to pass in a set of maps  (i.e. map<index, point-info>). We can then provide optimized back-end implementations based on:

1) Arrays for continuous indices
2) std::map for discontinuous mode

emgre, 2019-03-07:
See also #244.

jadamcrain, 2019-05-14:
If this is backed by std::map, you can use [lower_bound](https://en.cppreference.com/w/cpp/container/map/lower_bound) when searching the map using range.

Say a master asks for points 4 - 7 but only 5 - 8 are defined.  `lower_bound(4)` will return an iterator pointing to 5.

jadamcrain, 2019-06-06:
Resolved in [PR 296](https://github.com/dnp3/opendnp3/pull/296)

closed 2019-06-06 by jadamcrain