dnp3/opendnp3 #268: Master.getStatistics() returns null when using Java bindings

SecondFlight, 2019-05-10:
Adding the following code in `MasterDemo.java` after `master.enable()` (line 82) causes a `NullReferenceException`:

```java
System.out.println(master.getStatistics().link.numBadMasterBit);
```

The `NullRefererenceException` happens because `master.getStatistics().link` is null.

If my understanding is correct, the following C++ code should be equivalent:

```c++
std::cout << master->GetStackStatistics().link.numBadMasterBit << std::endl;
```

This C++ code works as expected.

jadamcrain, 2019-05-10:
Here's the JNI function that does the conversion:

https://github.com/automatak/dnp3/blob/2.0.x/java/cpp/adapters/Conversions.cpp#L22

I'll look into it during this release cycle.

emgre, 2019-05-13:
Thanks for the bug report. The issue was caused by a reference that was deleted in the native code.

Zannith, 2019-05-14:
Not sure if I should make another bug for this, but this same issue will pop up in com_automatak_dnp3_impl_ChannelImpl.cpp. Also as an effect of the test for this fix the maven build will fail because it is the only test which makes use of the dll and the java.library.path is not set for the maven-surefire-plugin. What I did to make it work for mine was add the argLine to the plugin configuration. Not sure if there is a more elegant way to set this?

```
<groupId>org.apache.maven.plugins</groupId>
<artifactId>maven-surefire-plugin</artifactId>
<version>${surefire-plugin.version}</version>
<configuration>
 <argLine>-Djava.library.path=C:\PathToRepo\build\Release</argLine>
 <includes>
   <include>**/*.class</include>
 </includes>
 <reportFormat>brief</reportFormat>
 <useFile>false</useFile>
</configuration>
```

emgre, 2019-05-14:
You're absolutely right about the `ChannelImpl`, I forgot it. Thanks!

As for the lib path, I'm not super comfortable in hardcoding a path in the POM. I took for granted that the `java.library.path` was properly set when we run the unit tests. Personally, I set it in the launch configuration of IntelliJ. I'll see with Adam what he thinks is best.

There is another test that uses the DLL/so (`DNP3ManagerIntegrationTest`), but it was ignored. I removed the `@Ignore`, because it makes sense to run it.

PR #271 (2019-05-13): Fixed Java getStatistics binding.
Fix #268.

The `getStatistics` for both the master and the outstation used to return a `LocalRef` that was deleted when the native call returned, causing undefined behaviour on the Java side. Now, the reference is promoted to a global reference that will be garbage collected by the JVM.

I also added a test case to reproduce the issue.
PR #273 (2019-05-14): Fix the IChannel::GetStatistics Java bindings.
Fix #268, I forgot `ChannelImpl` in in #271.
closed 2019-05-14 by jadamcrain