dnp3/opendnp3 #290: migrate library to use header-only Automatak libs

jadamcrain, 2019-05-24:
https://github.com/automatak/ser4cpp
(similar functionality now in openpal)

https://github.com/automatak/exe4cpp
(similar functionality now in openpal/asiopal)

https://github.com/automatak/log4cpp
(similar functionality now in openpal)

Additionally, regarding the public API:

* ser4cpp should be completely private (eventually). The octet strings currently use RSlice but we can make this a dirt simple buffer class `struct { pointer, length }` for the public API. The layer interfaces use this as well, but they'll be private once we smash the 4 separate libs down to a single library/namespace.

* exe4cpp should be completely private. DNP3Manager allocates the StrandExecutor internally.

* log4cpp could be private if we want to provide our own shim ILogHander/LogEntry? It might be worth it so that the public API can be completely free of log4cpp types.

emgre, 2019-06-07:
This is done in the 3.0.0 branch.
- The library is now a single target called `opendnp3`.
- `openpal` doesn't exist anymore.
- `ser4cpp`, `exe4cpp` and `log4cpp` are now used.
- `ser4cpp` and `exe4cpp` are completely private, so the user doesn't need them to build something.
- Hiding `log4cpp` from the public interface needs some more reflexion.

jadamcrain, 2019-06-11:
What do we lose by not hiding log4cpp::ILogHandler/etc? We can install it like any other header, but usually libraries like to install themselves.

emgre, 2020-04-27:
`log4cpp` was completely copied into opendnp3, so that we have no public dependencies.

PR #367 (2020-04-27): Fix remaining items for 3.0 release
- Fix #290: the only lib that was still exposed publicly was `log4cpp`. Since it's small and not really used elsewhere, it was copied right into opendnp3 and all references to it was removed. The library doesn't have any public dependency 🎉 (except for OpenSSL when compiling with TLS support).
- Fix #238: `TaskConfig` now takes a `std::shared_ptr` instead of a reference.
- Fix #293: Instead of having an out-parameter for the result of `DNP3Manager` calls, it now throws a `DNP3Error`.
  - All methods of `DNP3Manager` can throw an exception, notably because th [... cut]
closed 2020-04-27 by emgre