dnp3/opendnp3 #291: Class 2/3 scans not working

jadamcrain, 2019-05-27:
From a user:

"I have found the problem preparing a summary of my code in a simple class to show you an example. But anyway, only works in version 2.2.0. If you execute the outstation example of library code "https://github.com/dnp3/opendnp3/tree/2.0.x/dotnet/examples/outstation" with version 2.3.0, scan class from master only works for class0 and class1, not for class2 or class3."

emgre, 2019-06-21:
I confirm the issue with the C# bindings on 2.3.1. The `EventConfig::clazz` is just never set, so it always default itself to `PointClass::Class1`. Looking at 2.2.0, I doubt it ever worked properly. Fix is in progress.

emgre, 2019-06-21:
Fix is available in https://github.com/dnp3/opendnp3/commit/a8020d14a7dbf75672cbe823b65f7b640cf545b2.

PR #318 (2019-06-21): C# event class association (develop)
Fixes #291 in develop branch. It also fixes another issue introduced in the `develop` branch, where the `OutstationConfig` was not set properly.
PR #317 (2019-06-21): C# event class association
Fixes #291 in 2.x release branch.
closed 2019-06-21 by emgre