dnp3/opendnp3 #292: Issues with ISOEHandler

jadamcrain, 2019-05-31:
1) The interface can't tell you anything about which fragment the current one is in a multi-fragment response.

2) Since only a single instance is specified, it's impossible to figure out which request triggered the callback.

The resolution is to 1) extend the methods with the additional info needed and 2) pass in an instance on a per-request basis, initializing the master session with an instance for unsolicited responses. Note that 2) still allows you to use the same instance for all requests if you don't care.

jadamcrain, 2019-06-07:
Implemented 1) in [PR 298](https://github.com/dnp3/opendnp3/pull/298).

Now need to decide if 2) is worth it or not.

emgre, 2019-11-18:
2) was implemented in https://github.com/dnp3/opendnp3/pull/339 I think.

closed 2019-11-20 by emgre