dnp3/opendnp3 #293: Use exceptions instead of error_code

jadamcrain, 2019-05-31:
Opendnp3 **never** uses exceptions internally, but they make a lot of sense for parts of the public API, e.g. when you add a channel to the DNP3Manager with bad configuration.

blytkerchan, 2019-06-04:
FWIW The way I usually address this in public-facing APIs is by having everything internally not use exceptions, and defining the outer APIs as non-throwing with an  `std::nothrow_t const&` parameter and an overload without that parameter that does throw.

Example from RTIMDB:
```
#ifdef RTIMDB_ALLOW_EXCEPTIONS
	void sendCommand(unsigned int point_id, Details::CROB const &crob);
#endif
	Errors sendCommand(unsigned int point_id, Details::CROB const &crob RTIMDB_NOTHROW_PARAM);

```
In this case, when `RTIMDB_ALLOW_EXCEPTIONS` is defined, `RTIMDB_NOTHROW_PARAM` expands to `std::nothrow_t const& whereas if it's not defined, it expands to an empty string. This allows applications that don't use exceptions not to bother with putting std::nothrow in all their calls, and applications that do use exceptions keep the choice.

HTH

emgre, 2020-04-27:
Exceptions are now thrown only in `DNP3Manager` (so at startup). Also, we validate that we can bind to the port at creation of a channel and throw if we cannot (because it represents a misconfiguration).

commit 4da82d0a (2020-04-23): Use exceptions instead of error_code. Fix #293.
PR #367 (2020-04-27): Fix remaining items for 3.0 release
- Fix #290: the only lib that was still exposed publicly was `log4cpp`. Since it's small and not really used elsewhere, it was copied right into opendnp3 and all references to it was removed. The library doesn't have any public dependency 🎉 (except for OpenSSL when compiling with TLS support).
- Fix #238: `TaskConfig` now takes a `std::shared_ptr` instead of a reference.
- Fix #293: Instead of having an out-parameter for the result of `DNP3Manager` calls, it now throws a `DNP3Error`.
  - All methods of `DNP3Manager` can throw an exception, notably because th [... cut]
closed 2020-04-27 by emgre