dnp3/opendnp3 #294: Java Master demo is not getting updated timestamps

Zannith, 2019-06-03:
Using the Java master/outstation example I added the following lines after `outstation.enable();`: 
```
OutstationChangeSet set = new OutstationChangeSet();
set.update(new AnalogInput(10, (byte)1, System.currentTimeMillis()), 0);
outstation.apply(set);
```

Now when I startup the Master, when I get analog values it prints out `0 - Analog(10.000000, 0x1, 0)` I would expect that final piece to be a non-zero timestamp.

Using the equivalent cpp code with the UpdateBuilder and DNPTime the master example receives correctly updated timestamp

jadamcrain, 2019-06-03:
Thanks, we'll have a look.

Zannith, 2019-06-03:
Thank you, I looked into the source code some, but with my limited knowledge of the code I havn't come up with anything yet. I will continue to look and update here if I find anything.

emgre, 2019-06-07:
What group and variation are you requesting/receiving? The timestamp will only appear on analog input event (group 32) with a proper variation (var 3, 4, 7 or 8). Otherwise, the master won't have access to the timestamp.

I tried it using the master branch and the Java bindings with the following:
```
OutstationChangeSet set = new OutstationChangeSet();
set.update(new AnalogInput(10, (byte)1, System.currentTimeMillis()), 0);
outstation.apply(set);

master.scan(Header.allObjects((byte)32, (byte)3));
```
And it advertises the timestamp properly:
```
start asdu
HeaderInfo(Group32Var3, UINT16_CNT_UINT16_INDEX, timestamp: SYNCHRONIZED, isEvent: true, flagsValid: true, #: 0)
0 - Analog(10,000000, 0x1, 1559912984226)
end asdu
```

I also tried requesting a class 1 with the default variation of the analog input set to 3, and the timestamp is there.

Zannith, 2019-06-10:
Ah thank you emgre, The library does exactly as it should, the problem was a my own misunderstanding of DNP and the variation to use.

closed 2019-06-10 by Zannith