dnp3/opendnp3 #297: How to see the added DNP3 points in simulator

singha42, 2019-06-07:


singha42, 2019-06-07:
how to read the log files of the simulator

jadamcrain, 2019-06-07:
Which simulator are you referring to? The core library (this repository) has some master/outstation examples, but no "simulator".

singha42, 2019-06-07:
I was trying to establish a connection with a VM acting as outstation and automatak simulator as a master. So when the master connects I see in the outstation logs that it has added 864 points but in the attached logs, the value or name does not appear similar to the point added.

An example of the point :

2019-06-07 18:05:14,311:dnp3.outstation:DEBUG: Adding Point: PointDefinition ACLineSegment_l81_Voltage (30.3, index=15, type=Analog Input)

[image]

jadamcrain, 2019-06-08:
Two things:

1) The DNP3 protocol doesn't communicate the names of points. Each point is associated with an index in the range [0, 65535].

2) What do you read for "Analog" instead of "AnalogOutputStatus"?

If your device isn't configured for unsolicited reporting, you'll have to add a "scan" to see the data update.

singha42, 2019-06-10:
[image]

singha42, 2019-06-10:
Thank you for clarification. But I see in my outstation code, unsolicited requests is 'true'. 

event_buffers: (integer) Size of the database event buffers. Default: 10.
--
88 | allow_unsolicited: (boolean) Whether to allow unsolicited requests. Default: True.
89 | link_local_addr: (integer) Link layer local address. Default: 10.

jadamcrain, 2019-06-11:
I trust that those 5 values are the last ones that the simulated master saw from the 3rd party outstation.  

Can you provide a packet capture of the conversation using Wireshark?

closed 2019-06-17 by jadamcrain