dnp3/opendnp3 #307: run circle CI builds w/ additional CXX flags

jadamcrain, 2019-06-12:
To avoid warnings from piling up during development, let's get our CI builds using these options:

```
-DCMAKE_CXX_FLAGS="-Wall -Werror"
```

emgre, 2019-06-12:
It's done, but we exclude `-Wformat-truncation`, because the warnings are not important.

Do we want to add `-Wextra` ?

jadamcrain, 2019-06-13:
It depends if there are warnings that are actually useful

emgre, 2019-06-21:
`-Wextra` adds a couple of interesting warnings, most notably `sign-compare` and `type-limits`. But it also adds the very noisy and irrelevant `unused-parameter` that I manually disabled.

Added in https://github.com/dnp3/opendnp3/commit/d52c9103507fb93a270884309cc005b2cb25ddaa and https://github.com/dnp3/opendnp3/commit/908fcf060f1a806da2be0023c5cf9e2c6472ab4a.

PR #316 (2019-06-21): Support g51v2 (CTO Unsynchronized)
Closes #124.

- DNPTime now has a `TimestampQuality` attached to it. By default, it is set at `TimestampMode::INVALID`.
- When building `g2v3` and `g4v3` responses, the proper `g51` variation is used to according to the timestamp quality.
- `Flags` now has an explicit constructor. Otherwise, it led to wrong calls when updating binary input values (because `bool` is synonym of `uint8_t` on some platforms)
- Test procedure 8.15.2 of the certification procedures were executed to confirm proper handling (also added to dnp4s).
- Renamed IIN1.0 from `ALL_STATIONS`  [... cut]
closed 2019-06-21 by emgre