dnp3/opendnp3 #321: Unused unsolRetryTimeout parameter

emgre, 2019-07-04:
The `OutstationParams::unsolRetryTimeout` parameter seems to be unused. The retries are sent whenever the `OutstationParams::unsolConfirmTimeout` is expired. We could either remove that parameter entirely, or implement a retry delay. According to the spec (Figure 4-15, p.89), we should retry on every confirmation period, so adding a retry delay would apparently be against the spec.

jadamcrain, 2019-07-04:
An indefinite retry (same data  + seq) seems optimal to me. It ensures the master never sees duplicate event data and it's easy to implement... no additional state or counters.

My only concern here is the test procedures that don't seem to acknowledge that the specification allows for this.

emgre, 2019-07-05:
I will remove it in #322.

closed 2019-11-15 by emgre