dnp3/opendnp3 #325: Java bindings fail to build javadocs using Java 11

msqr, 2019-07-16:
I'm trying to build the release 2.3.2 Java bindings on Debian 10, which uses OpenJDK 11. The build fails with:

```
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-javadoc-plugin:2.9:jar (attach-javadocs) on project opendnp3-bindings: MavenReportException: Error while creating archive: 
[ERROR] Exit code: 1 - javadoc: error - The code being documented uses modules but the packages defined in http://docs.oracle.com/javase/7/docs/api/ are in the unnamed module.
[ERROR] 
[ERROR] Command line was: /usr/lib/jvm/java-11-openjdk-i386/bin/javadoc @options @packages
[ERROR] 
[ERROR] Refer to the generated Javadoc files in '/home/solar/dnp3/java/bindings/target/apidocs' dir.
```

I found that by adding an explicit Java source version of `1.8` to `pom.xml`, the build succeeds:

```xml
            <plugin>
                <groupId>org.apache.maven.plugins</groupId>
                <artifactId>maven-javadoc-plugin</artifactId>
                <version>${maven-javadoc-plugin.version}</version>
                <configuration>
                    <!-- Default configuration for all reports -->
                    <source>1.8</source>
                </configuration>
```

msqr, 2019-07-16:
Here's an exact diff of my fix:

```diff
diff --git a/java/pom.xml b/java/pom.xml
index fe7211a53..ad650da67 100644
--- a/java/pom.xml
+++ b/java/pom.xml
@@ -102,6 +102,7 @@
                 <version>${maven-javadoc-plugin.version}</version>
                 <configuration>
                     <!-- Default configuration for all reports -->
+                   <source>1.8</source>
                 </configuration>
                 <executions>
                     <execution>
```

jadamcrain, 2019-07-16:
thanks @msqr. We'll get this change into the next release.

emgre, 2019-11-18:
I'm adding `<source>8</source>` in the develop branch, but be aware that this was ruled as a bug in the JDK and it took a quite some time before the fix was back-ported to OpenJDK 11 (the version used in Debian Buster). See https://bugs.openjdk.java.net/browse/JDK-8219474 (the original issue is https://bugs.openjdk.java.net/browse/JDK-8212233).

emgre, 2020-01-17:
The source version is now specified since 625f90b838d3a48e1aaeecf49a4dfab78331d84d.

closed 2020-01-17 by emgre