dnp3/opendnp3 #326: Install not created

JakeGridEdge, 2019-07-17:
When following the build process on GitHub, following a make -j there is supposed to be a make install call. After using cmake and make, no install file was ever created in my library. Is there no install file anymore or is it somewhere other than the made dnp3 directory?

jadamcrain, 2019-07-19:
Version of opendnp3 and Linux platform? We'll give it a try.

xaoxtheory, 2019-07-22:
Ran into the same problem on Ubuntu 19.04 using the default branch (which is develop at the moment). Got everything working by checking out the release-2.x branch.

jadamcrain, 2019-07-22:
Yes, stick to a tagged release. Massive refactoring and changes taking place on develop, including to the CMake configuration.

closed 2019-07-22 by jadamcrain