dnp3/opendnp3 #328: Build on develop branch

jefim-indegy, 2019-08-01:
While performing Linux build according to instructions it fails (Ubuntu 18; cmake >= 3.12):

```
% sudo make install                                                                                                                                                                                                            
make: *** No rule to make target 'install'.  Stop.

jadamcrain, 2019-08-01:
The develop branch is for development. We've been refactoring the CMake build there, so nothing is stable.

Please use the latest tagged release:

https://github.com/dnp3/opendnp3/tree/2.3.2

closed 2019-08-01 by jadamcrain

emgre, 2019-08-01:
@jadamcrain Indeed. I changed it. I also added on my todo list the CMake install support for the `develop` branch.