dnp3/opendnp3 #329: Running Main Example

JakeGridEdge, 2019-08-01:
I am currently trying to make the master main.cpp example an executable in ./opendnp3/cpp/examples/master/main.cpp. When I call, sudo g++ main.cpp -o main on the file, I receive undefined references such as below:
`main.cpp:(.text+0x3b8): undefined reference to `openpal::TimeDuration::Seconds(long long)'
main.cpp:(.text+0x4bc): undefined reference to `opendnp3::ClassField::AllClasses()'
main.cpp:(.text+0x4d8): undefined reference to `openpal::TimeDuration::Minutes(long long)'

Is there a specified location where i should place this main to enable me to use it as an executable? Any help would be greatly appreciated!

jadamcrain, 2019-08-01:
You need to link the libraries:

https://dnp3.github.io/docs/guide/current/build/libraries/

The best way to see how to invoke the compiler is to run makefile generated by CMake w/ VERBOSE=1.

closed 2019-08-01 by jadamcrain

Marathonas, 2021-06-29:
Hello, I am having a similar issue as JakeGridEdge. I don't know much about all this stuff.
I ran the following commands on the Ubuntu 20.04 terminal:
   git clone https://github.com/dnp3/opendnp3
Entered the opendnp3 folder that was created, and ran:
   mkdir build
   cd build
   cmake .. -DDNP3_EXAMPLES=ON
   make -j VERBOSE=1
   sudo make install

After doing this I see some messages that make me think the outstation-demo is being installed. 
   "[ 99%] Linking CXX executable outstation-demo"
I uploaded a file with more message of the VERBOSE output.

Then I go to the path where the example is: ./opendnp3/cpp/examples/outstation and I try to run:
   sudo g++ main.cpp -o main
or
   sudo g++ main.cpp

And I get similar results as the ones above of undefined references. Here are a few of the messages:
/usr/bin/ld: /tmp/ccgO0KPD.o: in function `ConfigureDatabase()':
main.cpp:(.text+0x4c): undefined reference to `opendnp3::DatabaseConfig::DatabaseConfig(unsigned short)'
/usr/bin/ld: /tmp/ccgO0KPD.o: in function `main':
main.cpp:(.text+0x187): undefined reference to `opendnp3::DNP3Manager::DNP3Manager(unsigned int, std::shared_ptr<opendnp3::ILogHandler>, std::function<void (unsigned int)>, std::function<void (unsigned int)>)'
/usr/bin/ld: main.cpp:(.text+0x275): undefined reference to `opendnp3::DNP3Manager::AddTCPServer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, opendnp3::LogLevels const&, opendnp3::ServerAcceptMode, opendnp3::IPEndpoint const&, std::shared_ptr<opendnp3::IChannelListener>)'

What am I doing wrong? How can I get the example outstation code to run?
The link that was on the answer to be able to link the libraries was no available.

[opendnp3_ubuntu_install_log.txt](https://github.com/dnp3/opendnp3/files/6736449/opendnp3_ubuntu_install_log.txt)

emgre, 2021-06-30:
The examples are already built by CMake when you specify `DNP3_EXAMPLES=ON`. In your build directory, you will find a `outstation-demo` ready to run. You don't need to manually invoke GCC.

Your `sudo g++ main.cpp -o main` will certainly fail since you are not linking with OpenDNP3 whatsoever. I would encourage you to use CMake instead of building it by hand.

The link Adam was pointing to is roughly this: https://dnp3.github.io/docs/guide/3.0.0/build/cmake/

Marathonas, 2021-07-02:
Thanks for the reply! I managed to run the demo and I got the outstation to work!

I did it by running a basic script in the terminal of a text file containing "./outstation-demo" inside.
Is this what everyone does to run a script? Or is there another way to run it?