dnp3/opendnp3 #346: Linking issues with vcpkg generated build

OOPMan, 2019-12-12:
Hi, I'm looking to make use of opendnp3 in a project that uses `vcpkg`.

I've experimented with creating a vcpkg port that covers OpenDNP3 and while it seems to work correctly with regards to building the libraries I'm having issues actually linking the built libraries into my test project.

For reference, my vcpkg port can be found at https://github.com/OOPMan/vcpkg/tree/opendnp3

After installing the port I noticed that opendnp3 doesn't include any CMake targets for integration into an existing CMake project.

I got around this by adding the following to my CMakeLists.txt:

```
find_library(OPENPAL openpal)
find_library(OPENDNP3 opendnp3)
find_library(ASIOPAL asiopal)
find_library(ASIODNP3 asiodnp3)
target_link_libraries(test PUBLIC openpal)
target_link_libraries(test PUBLIC opendnp3)
target_link_libraries(test PUBLIC asiopal)
target_link_libraries(test PUBLIC asiodnp3)
```

This allows the headers to be found and the linked does process the libraries but I get errors of the form:

```
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(ConsoleLogger.cpp.o): in function `asiodnp3::ConsoleLogger::Log(openpal::LogEntry const&)':
ConsoleLogger.cpp:(.text+0x1bd): undefined reference to `opendnp3::LogFlagToString(int)'
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(DNP3ManagerImpl.cpp.o): in function `asiodnp3::DNP3ManagerImpl::Shutdown()':
DNP3ManagerImpl.cpp:(.text+0xd5): undefined reference to `asiopal::ResourceManager::Shutdown()'
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(DNP3ManagerImpl.cpp.o): in function `asiodnp3::DNP3ManagerImpl::DNP3ManagerImpl(unsigned int, std::shared_ptr<openpal::ILogHandler>, std::function<void ()>, std::function<void ()>)':
DNP3ManagerImpl.cpp:(.text+0x197): undefined reference to `openpal::Logger::Logger(std::shared_ptr<openpal::ILogHandler>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, openpal::LogFilters)'
/usr/bin/ld: DNP3ManagerImpl.cpp:(.text+0x529): undefined reference to `asiopal::ThreadPool::ThreadPool(openpal::Logger const&, std::shared_ptr<asiopal::IO> const&, unsigned int, std::function<void ()>, std::function<void ()>)'
/usr/bin/ld: DNP3ManagerImpl.cpp:(.text+0x5a7): undefined reference to `vtable for asiopal::ResourceManager'
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(DNP3ManagerImpl.cpp.o): in function `asiodnp3::DNP3ManagerImpl::~DNP3ManagerImpl()':
DNP3ManagerImpl.cpp:(.text+0x763): undefined reference to `asiopal::ThreadPool::~ThreadPool()'
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(DNP3ManagerImpl.cpp.o): in function `asiodnp3::DNP3ManagerImpl::AddTCPClient(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, asiopal::ChannelRetry const&, std::vector<asiopal::IPEndpoint, std::allocator<asiopal::IPEndpoint> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<asiodnp3::IChannelListener>)':
DNP3ManagerImpl.cpp:(.text+0x97f): undefined reference to `asiopal::Executor::Executor(std::shared_ptr<asiopal::IO> const&)'
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(DNP3ManagerImpl.cpp.o): in function `asiodnp3::DNP3ManagerImpl::AddTCPServer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, opendnp3::ServerAcceptMode, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short, std::shared_ptr<asiodnp3::IChannelListener>)':
DNP3ManagerImpl.cpp:(.text+0x11f2): undefined reference to `asiopal::Executor::Executor(std::shared_ptr<asiopal::IO> const&)'
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(DNP3ManagerImpl.cpp.o): in function `asiodnp3::DNP3ManagerImpl::AddSerial(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, asiopal::ChannelRetry const&, asiopal::SerialSettings, std::shared_ptr<asiodnp3::IChannelListener>)':
DNP3ManagerImpl.cpp:(.text+0x1abb): undefined reference to `asiopal::Executor::Executor(std::shared_ptr<asiopal::IO> const&)'
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(DNP3ManagerImpl.cpp.o): in function `asiodnp3::DNP3ManagerImpl::CreateListener(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, openpal::LogFilters, asiopal::IPEndpoint, std::shared_ptr<asiodnp3::IListenCallbacks> const&, std::error_code&)':
DNP3ManagerImpl.cpp:(.text+0x2168): undefined reference to `asiopal::Executor::Executor(std::shared_ptr<asiopal::IO> const&)'
/usr/bin/ld: DNP3ManagerImpl.cpp:(.text+0x24ca): undefined reference to `asiopal::TCPServer::StartAccept()'
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(DNP3ManagerImpl.cpp.o): in function `asiodnp3::DNP3ManagerImpl::CreateListener(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, openpal::LogFilters, asiopal::IPEndpoint, asiopal::TLSConfig const&, std::shared_ptr<asiodnp3::IListenCallbacks> const&, std::error_code&)':
DNP3ManagerImpl.cpp:(.text+0x27ca): undefined reference to `asiopal::Executor::Executor(std::shared_ptr<asiopal::IO> const&)'
/usr/bin/ld: DNP3ManagerImpl.cpp:(.text+0x2bcb): undefined reference to `asiopal::TLSServer::StartAccept(std::error_code&)'
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(DNP3ManagerImpl.cpp.o): in function `asiodnp3::DNP3ManagerImpl::AddTLSClient(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, asiopal::ChannelRetry const&, std::vector<asiopal::IPEndpoint, std::allocator<asiopal::IPEndpoint> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, asiopal::TLSConfig const&, std::shared_ptr<asiodnp3::IChannelListener>, std::error_code&)':
DNP3ManagerImpl.cpp:(.text+0x2f3e): undefined reference to `asiopal::Executor::Executor(std::shared_ptr<asiopal::IO> const&)'
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(DNP3ManagerImpl.cpp.o): in function `asiodnp3::DNP3ManagerImpl::AddTLSServer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, opendnp3::ServerAcceptMode, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short, asiopal::TLSConfig const&, std::shared_ptr<asiodnp3::IChannelListener>, std::error_code&)':
DNP3ManagerImpl.cpp:(.text+0x395e): undefined reference to `asiopal::Executor::Executor(std::shared_ptr<asiopal::IO> const&)'
/usr/bin/ld: /home/AJorgensen/Development/Tools/C++/vcpkg/installed/x64-linux/lib/libasiodnp3.a(DNP3ManagerImpl.cpp.o): in function `std::_Sp_counted_ptr<asio::ssl::detail::openssl_init_base::do_init*, (__gnu_cxx::_Lock_policy)2>::_M_dispose()':
DNP3ManagerImpl.cpp:(.text._ZNSt15_Sp_counted_ptrIPN4asio3ssl6detail17openssl_init_base7do_initELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv[_ZNSt15_Sp_counted_ptrIPN4asio3ssl6detail17openssl_init_base7do_initELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv]+0x1e): undefined reference to `CRYPTO_set_locking_callback'
/usr/bin/ld: DNP3ManagerImpl.cpp:(.text._ZNSt15_Sp_counted_ptrIPN4asio3ssl6detail17openssl_init_base7do_initELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv[_ZNSt15_Sp_counted_ptrIPN4asio3ssl6detail17openssl_init_base7do_initELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv]+0x23): undefined reference to `ERR_free_strings'
/usr/bin/ld: DNP3ManagerImpl.cpp:(.text._ZNSt15_Sp_counted_ptrIPN4asio3ssl6detail17openssl_init_base7do_initELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv[_ZNSt15_Sp_counted_ptrIPN4asio3ssl6detail17openssl_init_base7do_initELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv]+0x28): undefined reference to `EVP_cleanup'
/usr/bin/ld: DNP3ManagerImpl.cpp:(.text._ZNSt15_Sp_counted_ptrIPN4asio3ssl6detail17openssl_init_base7do_initELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv[_ZNSt15_Sp_counted_ptrIPN4asio3ssl6detail17openssl_init_base7do_initELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv]+0x2d): undefined reference to `CRYPTO_cleanup_all_ex_data'
[... cut ...] Writer&)#1}>::_M_invoke(std::_Any_data const&, opendnp3::HeaderWriter&)':
Conversions.cpp:(.text+0x30): undefined reference to `opendnp3::Header::WriteTo(opendnp3::HeaderWriter&) const'
collect2: error: ld returned 1 exit status
make[2]: *** [apps/CMakeFiles/opendnp3_rest_server.dir/build.make:88: apps/opendnp3_rest_server] Error 1
make[1]: *** [CMakeFiles/Makefile2:926: apps/CMakeFiles/opendnp3_rest_server.dir/all] Error 2
make: *** [Makefile:95: all] Error 2
```

Any idea what's going on here?

jadamcrain, 2019-12-12:
We'll take a look at the details. Be aware that in the 3.0 release we've been working on, there will only be a single library which will likely make things a lot easier for you.

OOPMan, 2019-12-12:
That's great to hear. I managed to sort my problems out by consulting one of your other example projects and fiddling. For reference:

```
# OpenDNP3
find_package(OpenSSL REQUIRED)
find_library(OPENPAL openpal)
find_library(OPENDNP3 opendnp3)
find_library(ASIOPAL asiopal)
find_library(ASIODNP3 asiodnp3)
target_link_libraries(test LINK_PUBLIC asiodnp3 opendnp3 asiopal openpal OpenSSL::SSL OpenSSL::Crypto)
```

Is there an ETA on 3.0?

jadamcrain, 2019-12-12:
I'm hoping that it doesn't take longer than end of February.

OOPMan, 2019-12-20:
Going to mark this as closed as MS have merged by vcpkg port for opendnp3 :-)

closed 2019-12-20 by OOPMan

jadamcrain, 2019-12-20:
Thanks for your work making the project available via vcpkg